Ultimaker/Cura #13144: The slicer incorrectly placed the supports on some elements.

boris098765, 2022-08-28:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

FlyingBear Ghost 5 + Direct by SibMaker

### Reproduction steps

I placed a model of a photopolymer drain funnel in the slicer.

### Actual results

The slicer incorrectly placed the supports on some elements. Obviously, the slicer does not know how to identify islands (parts of the model printed in the air).

### Expected results

The slicer had to add support for each of the overhanging elements.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[2022-08-28_17-33-05.zip](https://github.com/Ultimaker/Cura/files/9439876/2022-08-28_17-33-05.zip)

GregValiant, 2022-08-30:
Thanks for the report.
You indicate in your settings that the Minimum Support Area should be 18mm² and that the Minimum Support Interface Area should be 10mm².  So Cura is building support according to your input.
If you want all those islands supported then you would need to drop the Minimum Support Area to a number smaller than the area of the bottom surfaces of those overhangs.
If you want support interfaces to be built on those supports you would also need to input a smaller number for Minimum Support Interface Area.  When you do both of those you get this structure which will support all of the overhanging features.
[image]

This appears to be a settings issue rather than a bug and so I will remove the "bug" label.  I'll leave it open if you wish to respond or if you have further information to provide.

boris098765, 2022-08-30:
Thank you for your help! This parameter was hidden by the settings.
And looking at what happened to my printer after trying to print this model with retracts (the plastic was erased and forgotten), I thought about the possibility of disabling rollbacks inside the support structures, but without losing rollbacks on the model. Is there such a possibility in Cura?

GregValiant, 2022-08-30:
There are two things I always do when starting on a model.
I re-select my profile and when Cura asks - I discard my changes.  That puts the profile back to a starting point.
Next, I use the Settings Visibility button (to the right of the Settings Search box) and set the visibility to "All".
In the long run those two things can save a lot of time.

In regards to your other question...I don't know what you mean by a "rollback".  Are you talking about the Support Infill?

boris098765, 2022-08-31:
I'm sorry, translation error. I'm talking about retracts.

GregValiant, 2022-08-31:
I don't think you can do it.  There are no "Travel" settings available as Per Model settings.  That means adjusting a Support Blocker to keep retractions from happening just in the support is not possible.
What you can do is enable "Limit Support Retractions" in the Travel Settings and in the Support Settings enable "Connect Support Lines".  Both "Connect Support Lines" (and "Connect Infill Lines") will limit the number of retractions as the extruder keeps running to connect the lines rather than stopping and starting constantly.

boris098765, 2022-09-05:
Thank you for your help! You have helped me a lot!

GregValiant, 2022-09-05:
I just sit here and type stuff.  Good job fixing the problem.
I'll go ahead and close this then.  If you think it should remain open you can still leave a message here.

closed 2022-09-05 by GregValiant