Ultimaker/Cura #13147: Cura5.1 refuses to use relative extrusion

AlexTOOT, 2022-08-29:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

custom made

### Reproduction steps

I got a printer running Reprap Firmware, using cura 5.1.0 looking forward to trying out the new adaptive infill feature
And for some reason, I have to use relative extrusion

### Actual results

Cura insists to putting M82 (absolute extrusion) into Gcode no matter I check the relative extrusion box or not. 
That's causing me tons of problem. Any fix?

### Expected results

Should insert M82 for absolute extrusion and M83 for relative extrusion.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9441637/cura.log)
[CFFFP_remote.zip](https://github.com/Ultimaker/Cura/files/9441642/CFFFP_remote.zip)

GregValiant, 2022-08-30:
Thanks for the report.
I've never had the problem myself and slicing your project file the gcode comes out with the M83 line immediately following your StartUp Gcode.
M291 P"Printing started !" R"Printing.." S1 T3
**M83 ;relative extrusion mode**
G1 F1800 E-2

So I am unable to duplicate this.  Maybe someone else has an idea of what could be going on.
Have you tried re-installing Cura?

AlexTOOT, 2022-08-31:
Actually, I haven't before

After re-installing, the problem is magically fixed
Sorry, should have tried that. Thanks for your time.

closed 2022-08-31 by AlexTOOT