Ultimaker/Cura #13148: Tree support mid-air

Ppieter21, 2022-08-29:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

I sliced a model that is basically a hollow cylinder and enabled tree support, with a branch angle of 40 degrees.(The same happens at 50 and 30 degrees)

### Actual results

[image]
As the tree supports got wider at the top to support the flat top of the cylinder, there would be a few places where the support wants to print mid-air.
[image]
This is an image with the Tree support branch angle set at 30 degrees.

### Expected results

The support should go wider without placing some of the branches mid-air.   All the branches should come off another branch, the model or the build plate.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Bug report project.zip](https://github.com/Ultimaker/Cura/files/9442159/Bug.report.project.zip)

GregValiant, 2022-08-30:
Thanks for the report.
I have noticed this behavior before and usually have been able to tune it out by playing with the various Tree support settings.  Someone from the Cura team may want to revisit this to insure that it can be "tuned out".

Moving away from the bug report for a minute...Looking at your settings I think there is room for adjustments that would allow you to have a support structure that would do what you need.
I think going back to conventional supports, at 10% density, 60% interface density, .4 horizontal expansion and .4 support interface horizontal expansion with the Minimum Support Area at 1.0 would do the job.

I think this is the tough area of that model.  You can see that the initial extrusion of the "split circle" is well supported.  As the print built up to the top of the pocket the roof would be well supported.
(BTW you have Pause at Height enabled at Layer 700 and the model isn't that tall.  I assume you had it enabled for a different model and haven't turned it off.)
[image]

Here is a view using your tree supports and settings.
[image]

DerGenaue, 2022-09-25:
Duplicate of #12326