Ultimaker/Cura #13154: Cura remains slicing

jfbelloc, 2022-08-29:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Geeetech A10

### Reproduction steps

Try to slice a piece, it remains slicing until I cancel. When cancelling, error message appears.
If I try to slice again directly goes to error message.

It was working fine yesterday with all pieces I'm trying.

Some days ago I had the same problem, I uninstalled Cura and cleaned registry, After installing again problem was solved, but after three or four days it's happening again.

I'm attaching two simple pieces, but it's happening with all of them.

### Actual results

First time it remains slicing until I cancel, next times error message appears immediatelly.

### Expected results

Slicing should finish.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9445769/cura.log)
[Pieces.zip](https://github.com/Ultimaker/Cura/files/9445796/Pieces.zip)

GregValiant, 2022-08-31:
Thanks for the report.
I was able to slice the models separately, and with both loaded at the same time.  With only the STL files to work with we only know that the models are OK.  A project file can tell us if there are conflicts in your settings that are causing slicing to fail.
Load a model and set Cura up to slice and then use the "File | Save Project" command.  Zip the 3mf file and post it here.  Without a project file I'm unable to duplicate the problem.
Someone from the Cura team will take a look.
[image]

jfbelloc, 2022-09-03:
Hello, thanks for your help.

I uninstalled Cura and installed it again. During two days it worked fine, but today I have the same problem again. Cura is not slicing the same piece. I'm attaching the same piece file, the today's log file and the project file.
[errorCura.zip](https://github.com/Ultimaker/Cura/files/9482291/errorCura.zip)

GregValiant, 2022-09-03:
Thanks for the file.
Once again It sliced fine for me.  I'm also on Windows 10.
There have been some issues with models not slicing when Win10 is the operating system.
Have you tried moving the model and rotating it?  There is a persistent bug report where the model won't slice when oriented at the build plate mid-point but when moved and/or rotated it's OK.  It doesn't seem to be specific to a printer model but all the reports have been for Windows computers.  The developers have been working on the problem.

jfbelloc, 2022-09-10:
Thanks for your help Greg, I'll have to wait for another Cura version.

closed 2022-09-10 by jfbelloc

GregValiant, 2022-09-10:
You could install 4.13.  It shouldn't interfere with 5.1.