Ultimaker/Cura #13155: 5.1 freezes at 99% at the preview stage

Scage75, 2022-08-29:
### Application Version

5.1

### Platform

windows 10

### Printer

snapmaker 2

### Reproduction steps

1. sliced model
2. clicked the preview button

### Actual results

freezes at 99%

### Expected results

should of opened up to view the supports and all

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

this seems to happen when model takes 3 hrs or longer to print

GregValiant, 2022-08-31:
Thanks for the report.
We need a project file in order to investigate.
Load your model, set Cura up ready to slice, and then select the "File | Save Project" command. Zip the3mf file and post the zip file here.

Scage75, 2022-08-31:
[S2A_supports.zip](https://github.com/Ultimaker/Cura/files/9465712/S2A_supports.zip)
here you go GregValiant. Im not having slicing issues just being able to preview after slicing.

GregValiant, 2022-09-02:
Well I can't check the preview because I can't get it to slice.  No error, it just won't finish and then the button goes back to "Slice".
The Cura Team will take a look.

Scage75, 2022-09-03:
that has happened tome before too on other parts.

MariMakes, 2022-11-08:
Hey @Scage75,

I was able to slice your printjob in Cura 5.2.1.
This is a huge printjob! It takes 9 days to complete. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

Scage75, 2022-11-08:
Yeah it can be closed. I sliced it up more so it won't take as long to print
Thanks for the help

<https://user-images.githubusercontent.com/40423138/200613659-6774f516-d2b3-469e-b445-6b60b6c6472d.png>

closed 2022-11-09 by MariMakes