Ultimaker/Cura #13156: error slicing file

mjohnson07, 2022-08-29:
### Application Version

5.1.0

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. trying to slice file at "fast" resolution
2. previously this file did slice at "normal" resolution
3. restarted computer, tried again at "fast" and it still failed

### Actual results

slicing fails

### Expected results

produce a sliced file

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[20220808_AC Plug Box.zip](https://github.com/Ultimaker/Cura/files/9446562/20220808_AC.Plug.Box.zip)

jjj03, 2022-08-30:
I cannot slice anything in cura 5.1.0. I get an error every time no matter the file.

MariMakes, 2022-09-01:
@mjohnson07,

Welcome to the Ultimaker Cura Github 🎉.

Thanks for the report. That's such a cool printjob!
I'm seeing the same error if I slice in Fast, but not if I slice in Normal. 
[image]

I'll bring it up to the team to be analyzed. 🤞

MariMakes, 2022-09-01:
@jjj03 Can you please make a seperate issue for what you are observing?
It seems different than the reported issue. 

You can click here to make a ticket: https://github.com/Ultimaker/Cura/issues/new/choose

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker