Ultimaker/Cura #13158: "Update profile from current settings"...doesn't

daniels220, 2022-08-30:
### Application Version

5.1.0

### Platform

macOS Big Sur

### Printer

Creality CR-10S

### Reproduction steps

Starting with an existing custom profile, tweak some settings and choose "Update profile with current settings/overrides". All the reset-to-profile arrows go away, so it looks like the profile has indeed been updated. Restart Cura.

### Actual results

The profile is in fact unchanged—the profile shows as unmodified, but the settings are also back to the way the profile was before you did anything.

### Expected results

The profile should actually be updated and the settings should remain as they were saved.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9449158/cura.log)

MariMakes, 2022-09-01:
Hey @daniels220,

Thanks for your report 👍 

I'm trying to troubleshoot your issue but I'm having some trouble reproducing your findings.
At first, I thought it was because I work on a Windows device, but I asked someone on a Mac device and they also could not reproduce the issue.

It's hard to resolve issues that we cannot reproduce. Are these the correct steps?
https://user-images.githubusercontent.com/40423138/187955990-7a99de49-1134-441c-a0d5-034156a0118a.mp4

Do you mind making a screen recording from what you are encountering?

daniels220, 2022-09-02:
Your steps are correct—actually the step at the end of flipping to another profile and back is not even necessary, in my case the issue is visible the moment I open Cura. Yes, I can make a screen recording:

https://user-images.githubusercontent.com/1410866/188202540-62f5b6ae-9e74-4365-8b48-2dd2caff63fe.mov

MariMakes, 2022-10-05:
We've had another report of this exact issue in Arachne Beta 2. 
https://youtu.be/cdTXoMoZgT0

I'll bring it up to the team to discuss what we can do. 
Thanks for the reports 👍

Dougoll, 2022-11-17:
I just had the same issues as him...I made tests all day and tweaked my profile. After a Cura restart, the update button did not save anything. Back to square one....really frustrating.  Using Cura 5.2.1 on Windows 11 with latest updates.

N6REJ, 2023-04-03:
just for the info, the latest cura 5.3.0 doesn't have the issue.  The "NEW" saving system works fine.  It's 5.3.0-alpha-xmas that has the issue.

daniels220, 2023-04-05:
Oh, that's wonderful. I'm certainly liking the indication of what has changed when saving a profile—that's been a pain point for me the whole time I've been using Cura. If this involved enough of a rewrite to fix this bug along the way, even better.