Ultimaker/Cura #13159: Performance regression updating settings display

daniels220, 2022-08-30:
### Application Version

5.1.0

### Platform

macOS Big Sur

### Printer

Creality CR-10S

### Reproduction steps

Just opening Cura, or searching in the settings search box, or to a very small extent just expanding a section of settings.

### Actual results

Settings are added-removed in chunks, with variable lag between each chunk, adding up to several seconds of unresponsiveness.

### Expected results

In Cura 4.x, and I believe the Arachne betas, there may have been a slight all-at-once lag, but then everything changed in a single screen refresh, and the process was overall much faster.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/1410866/187338021-ca826868-289e-4896-9405-08b437eb35d4.mov

MariMakes, 2022-09-01:
Hey @daniels220,

Thanks for your report.
I think I understand what you are trying to report. 🤔 
But when I'm comparing 4.13 and 5.1 I am not seeing a very noticeable difference. 
Especially once I use a search term I've used in the same session. 

I suspect that this issue will not get priority over the other issues we are currently trying to resolve. 😞 
Can you share in what way this inconveniences you? 

Thank you 😄

daniels220, 2022-09-02:
I definitely am still seeing a difference. Here for reference are Cura 4.13:

https://user-images.githubusercontent.com/1410866/188200958-63b77941-160f-4f25-a640-df2f7094f3c2.mov

Arachne beta 2:

https://user-images.githubusercontent.com/1410866/188200976-48008b0a-6110-4151-8a49-8c1806ce11b1.mov

and Cura 5.1 with the same search term for consistency:

https://user-images.githubusercontent.com/1410866/188201030-ad3b1887-4cc0-4701-bca3-eb7ca5bed3da.mov

The inconvenience...I mean, in the grand scheme of things it's minor, sure, but it adds up when it slows me down by a second or two every time I search for a setting—which is pretty often, because there are a _lot_ of settings (which is a good thing!) and hunting through them all is often pretty slow (especially if I don't know which _section_ the setting I want is in, only what it's called). It also just feels _glitchy_, which is never a good look.

daniels220, 2022-09-02:
Ah, this is actually effectively a duplicate of #13098, whose author goes into more technical detail of what is probably wrong and how it might be fixed. Feel free to close this as a duplicate, but if you do, please make sure to assign the "Bug" label to #13098 as the author there doesn't seem to have been able to do so.