Ultimaker/Cura #13162: Spirel makes it crash

imWACC0, 2022-08-30:
### Application Version

4.13.1

### Platform

Linux Mint

### Printer

Ender 3

### Reproduction steps

Download file 130 @ https://www.thingiverse.com/thing:3565827
Open file, try to manipulate (resize, rotate...)

### Actual results

Program crash

### Expected results

object to move/resize

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9452695/cura.log)

imWACC0, 2022-08-30:
Let's change that, EVERYTHING makes it crash.
I know it was working last week, I looked at a R-Pi case and an Eco Dot stand. Now those .STL files make it crash.
I tried the "reset" by deleting the Cura file, still not working.

MariMakes, 2022-09-01:
Hey @imWACC0,

Welcome to the Ultimaker Cura Github! 🚀 

Before we start troubleshooting, can you take a look at this comment?
I'm seeing some QT errors in your logs that might be related. 
https://github.com/Ultimaker/Cura/issues/12343#issuecomment-1161683871

Also we released two linux builds of Cura 5.1 to resolve some of the reports we've been having.
https://github.com/Ultimaker/Cura/releases/tag/5.1.0

Can you take a look to see if this works for you?

Thank you 😄

imWACC0, 2022-09-02:
Sorry, having a bad week. I hope my inaction is not messing anything for you

MariMakes, 2022-10-04:
Hey, @imWACC0 did you have a chance to look at this yet?

closed 2022-10-19 by github-actions[bot]