Ultimaker/Cura #13171: Slicing failed

elizabethtsds, 2022-08-31:
### Application Version

5.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Inserted 3d model to print, sliced and it showed an error

### Actual results

stops

### Expected results

Sent me to this page

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_vaults test.zip](https://github.com/Ultimaker/Cura/files/9459294/UMS5_vaults.test.zip)

GregValiant, 2022-08-31:
Thanks for the report.
As a workaround you can set the Layer Height to .11.  I got it to slice that way.  (I left the model vertical.)
Someone from the Cura team will take a close look.  I believe this behavior is being worked on by the Cura Team.

Piezoid, 2022-08-31:
For developers: This dereferences a null pointer in `SkeletalTrapezoidationGraph::collapseSmallEdges` at `quad_end->twin->twin = quad_start->twin;`, because `quad_end->twin = nullptr`.
[Increasing the integer precision from 1 µm to 0.1 µm](https://github.com/Ultimaker/CuraEngine/pull/1710) allows the model to slice:
[image]

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue:
[CURAENGINE-1P](https://ultimaker-o7.sentry.io/issues/4855795727/)