Ultimaker/Cura #13174: Slicing cancels during larger slicing attempts

jmmoore98, 2022-08-31:
### Application Version

5.1.0

### Platform

Windows 11; Evga GTX 1080; Ryzen 5 5600x

### Printer

Ender-3

### Reproduction steps

1. Opened Oggy Boogy model
2. Selected Dynamic Quality with a 20% infill
3. Start slice

### Actual results

Slicing fails with an error thrown

### Expected results

A fully sliced model that honestly I wasn't expecting to be able to print, but was expecting to wait for it to slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9461091/cura.log)
[Oggy.zip](https://github.com/Ultimaker/Cura/files/9461094/Oggy.zip)

This maybe a duplicate, but all the issues I saw were lacking in details as to when they were having issues (may have been poor searching on my part though)

GregValiant, 2022-08-31:
Thanks for the report.
I didn't make any changes to the project and it sliced for me so I'm unable to duplicate the problem.  The model doesn't seem to have any errors.
Someone from the Cura team will take a look at this.  In the meantime, try moving the model on the build plate and/or rotating the model.  This bug can sometimes be fooled.

jmmoore98, 2022-09-01:
I am leaning towards user error on my part. I didn't realize that I had other instances of Cura open. I wasn't having issues with other models, but closing those instances (with different models) did the trick. Maybe the size of the project when trying to slice brought it up to some memory limit? I'll leave this open in the off chance that someone on the Cura team is able to reproduce this.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker