Ultimaker/Cura #13176: Crash after Printer selection

Clem-, 2022-08-31:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

1. Fresh install and run Cura 5.1.0
2. Skip Cloud Login
3. Select Printer

### Actual results

Cura crashes

### Expected results

Run Cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9461530/cura.log)

GregValiant, 2022-08-31:
Thanks for the report.
I would get this problem occasionally with previous versions but I haven't had the problem with 5.x.
Someone from the Cura team will check your log file.

Clem-, 2022-09-21:
Hi, any updates ?
Still crashing in 5.1.1. 
Cannot use CURA at all.