Ultimaker/Cura #13180: Ender 3v2

narnoldus, 2022-08-31:
### Application Version

5.1.0

### Platform

windows 10

### Printer

Creality Ender 3v2 

### Reproduction steps

after updating Cura to version 5.1.0 after I have pre heated my printer I go to start the print and the nozzle moves to the front left corner and sits on the bed for a few seconds before it prints even though my printer temps are already heated.  another issue I have come across is the temp wanting to lower to 175 degrees after I start my print, so I have to change it to 200 degrees manually. 

### Actual results

I have tried updating my printer firmware and does not fix this issue I have also made sure all my temps in the Cura software are set to my printing temp.

### Expected results

it should start printing right away like it used to before the 5.1.0 update. and not drop the nozzle temp to 200 degrees.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

GregValiant, 2022-09-01:
Thanks for the report.
We need a project file in order to investigate.
Load any model (Benchy or calibration cube will work), set Cura up ready to slice, and then select the "File | Save Project" command. Zip the3mf file and post the zip file here.

Asterchades, 2022-09-01:
The Ender-3 Pro start GCode was modified in December last year to pre-heat the nozzle simultaneously with the bed. This modification heats to the material's listed Standby Temperature, not its first layer printing temperature. When the bed has finished heating it will then continue heating the nozzle to the specified initial layer temperature.

In other words, it's doing what it's being told to do as a result of having chosen that printer definition (which presumably you chose to avoid the excluded areas of the base Ender-3 profile). If you leave the temperature alone it will automatically correct it without your intervention, and it's sitting in the corner due to the need to restore its temperature after the lower standby temperature has been set (Marlin has no "Only set to this if not already higher" temperature commands).

You can either remove/comment out the line in the Start GCode which includes the {material_standby_temperature} placeholder to stop both elements heating at once, or change the placeholder to instead be {material_print_temperature_layer_0} to maintain simultaneous heating while also setting the pre-heat point more appropriate for an immediate start.

MariMakes, 2022-11-10:
Hey @narnoldus,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

To me, it doesn't seem like a bug in Ultimaker Cura anymore.
Especially with the suggestion from @Asterchades to change the start-gcode. 

Would you still like to keep this issue open or do you agree that we can close it?

narnoldus, 2022-11-10:
Haven’t had further issues thanks.

closed 2022-11-11 by MariMakes