Ultimaker/Cura #13182: Update Cura using AppImage Update tool

leoheck, 2022-09-01:
### Is your feature request related to a problem?

no

### Describe the solution you'd like

update app image with the tool to update appimages

### Describe alternatives you've considered

-

### Affected users and/or printers

-

### Additional information & file uploads

https://github.com/AppImage/AppImageUpdate

[image]

MariMakes, 2022-11-01:
Hey @leoheck 

Sorry, it took us a while to get back to you 😞 

Personally, I'm not a Linux user so I'm unaware of the added benefit of this feature request. 
So I'll have to bring it up with the team to discuss.

jellespijker, 2022-12-12:
Hi @leoheck sorry it took a bit longer to look into this. We're currently not equipped to distribute Cura (for any OS) in an update process. I also think that the benefit at this time for implementing this is limited and if we decide to follow such a distribution process it should be done for all three supported OSes. Which isn't in our planning currently.

I will keep the AppImage Update tool in the back of my mind if we ever decide to redesign our distribution process but for now I will defer this issue.

closed 2022-12-12 by jellespijker

leoheck, 2022-12-12:
_Disgusting_

[image]

leoheck, 2022-12-12:
Oh, thanks for the note. Which is better then just closing the issue without saying a word.

Now, regarding the OS completeness. It would be good to do for all, for sure, but the progress starts with a step. I invite the team to challenge yourselves by doing this step for linux with App Image update tool since it looks like it is the easiest step. 

Cheers.