Ultimaker/Cura #13184: Concentric skin and infill patterns do not respect wall distribution count

daniels220, 2022-09-01:
### Application Version

5.1.0

### Platform

macOS Big Sur

### Printer

Creality CR-10S

### Reproduction steps

Use concentric skin or 100%-infill patterns in a part with smoothly-varying dimensions.

### Actual results

All size deviation is always applied to the innermost line or pair of lines.

### Expected results

Distribution across many lines according to the wall line distribution count setting, similar to if I instead set a large wall line count. (This was the behavior under Arachne beta 2.) Or, a separate setting to control distribution for skin and/or infill would offer even more flexibility.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Same models as the bug with wall distribution count I just reported, but now with only the outer wall and 100% infill: [Distribution count not respected.3mf.zip](https://github.com/Ultimaker/Cura/files/9466930/Distribution.count.not.respected.3mf.zip)

MariMakes, 2022-09-02:
Hey @daniels220,

Thanks for your report. 👍 

I'm not sure if I understand your issue. 🤔 
You see that the variable linewidth is only applied to the inner 1 or 2 lines but you expected them to be distributed over more than 2 lines. 

But I noticed that you do not have enough walls where the distribution count can be applied to, you have only 1. If you bring it up to 5 than you can see the distribution over more than 2 lines. 
[image]

Here is a video of what I'm observing. 
https://user-images.githubusercontent.com/40423138/188139220-7211341e-ec55-433e-ad1a-0df1f86cdad6.mp4

Is this the behavior you are looking for or is your bug something else?
Can you perhaps add a drawing or a different model to illustrate your issue?

daniels220, 2022-09-02:
Right, so the whole point here is that, although they are technically not "walls", 100% concentric infill or skin is exactly _like_ walls in every way except which line-type-specific settings (speed, width, accel/jerk, etc) are applied. In Arachne Engine beta 2, such concentric patterns used the exact same layout algorithm as walls (AFAICT), including respecting the wall distribution count, so it was possible to have concentric infill or skin that evenly distributed variations across many lines. Here's Cura 5.1, similar to your video (I tweaked the minimum wall line width as well to make it more visible):

[image]

And Arachne with identical settings:

[image]

For whatever reason, in the final release of Cura 5, this capability has been removed. Admittedly this is a grey area between "bug" and "enhancement". I went with bug because (a) the behavior I want is not new—as I say, Arachne beta 2 already did the (IMO) right thing—and (b) AFAICT there is really no _downside_ to this behavior, so it should be fine to make it the default and only add separate settings to control it (which certainly would qualify as an enhancement) later if there is enough demand.

As far as why I consider this behavior superior—I mean, the bottom line is why _not_ distribute variance over many lines? But specifically I have used it to allow me to print an irregular solid part with very few line-count transitions, and therefore very few travel moves and very few visible artifacts when printing with clear filament:

[image]

Does that make sense?

GregValiant, 2022-09-04:
Just FYI
There have been reports by users having trouble with the Concentric pattern whether it is used for Ironing, Infill, Support Infill, Top/Bottom layers, everywhere it is used seems to be having some issues.
So my thought is that:  although in this instance the problem appears to be the way the line width is being handled, that it might be something more basic going on with all things "Concentric" and this behavior might be a symptom rather than the actual bug.

daniels220, 2022-09-04:
Interesting to know, but I find that pretty unlikely, since the behavior is identical to walls with Wall Distribution Count = 1. It really does seem like at some point it just stopped respecting that setting, and while I can see how it would be even _better_ to have a _separate_ setting for "wall" distribution count as applied to skin and infill, the Arachne beta behavior of using the one setting for everything was better than just having no control _at all._

nallath, 2022-09-09:
I'm fairly sure that we never applied Arachne on concentric infill. We only ever apply it to walls. Arachne Engine beta 2 used to have some extra strategies that have been removed as they decreased the quality in almost all of the cases (and were only there as a leftover from comparing the new algorithm with other algorithms in order to write a paper)

daniels220, 2022-09-10:
@nallath I mean, empirically you did? See the last screenshot in my original report. Or is that the result of one of those extra strategies somehow? It sure looks identical to the behavior if the number-of-walls setting is set really high... In any case, I think it would be extremely _useful_ to have the ability to apply the wall distribution strategy to any concentric pattern.

brewfest, 2025-04-10:
Writing to say this is still an issue, and that I as well would like a solution to. I use 100% concentric infill for most small parts as it keeps travels to a minimum and thus prevents stringing and other travel-related issues. Cura (now version 5.10) will introduce line-width issues when doing this, meaning it just trades one issue type for another. This is frustrating because it doesn't have to be this way, and could essentially be a free quality upgrade if it was changed to how @daniels220 suggested. 

As it currently stands, to fit the lines in 100% infill, only the innermost lines' widths' are dynamic. As shown, this clearly isn't the ideal performance. 

zero walls
[image]

one wall
[image]

three walls
[image]

The only current "solution" to increase the number of dynamic-width "walls" in the infill is to use only (real) walls, as any amount less forces the innermost area to be filled with non-ideal line widths infill.

99 walls
[image]

As it can be seen, printing only walls changes the line width variability from 0.30mm - 0.80mm down to 0.37mm - 0.53mm, a fairly large positive change for a 0.4mm nozzle I would say. Great, but only printing walls completely fails at reducing the number of travels!!! Even ignoring the issue of the innermost infill lines being unideal widths, they often dead end to fit into the remaining space, thus causing a travel, defeating the purpose of a contiguous infill itself! Argh!

Having a setting just like "Wall Distribution Count" for infill, except starting from the center rather than the outside would resolve this.  Please, this setting would be a godsend for anyone who prints with stringy oozy materials like Polycarbonate where travels severely diminish the quality of the print.

brewfest, 2025-07-25:
Found some evidence today concentric infill ignoring wall distribution count is indeed bugged and not functioning as intended. In plain writing, the tooltip literally states "This setting doesn't just apply to normal walls, but also to (....) concentric patterns"

[image]

As a side issue, it's frustrating printing a part that features details with only a single inner and outer wall and Cura black box decides to ignore the outer wall to distribute the line width to. Let me decide if the outer wall should be included or not, or even give me a seperate Minimum Wall Line Width setting just for the out wall.

GregValiant, 2025-07-25:
@HellAholic this one has been brought back to life.  Do you have anything to add here?

HellAholic, 2025-07-25:
It seems more appropriate to treat this as a feature request—to add a setting that allows users to control the line width for infill patterns, such as concentric. I'm not sure how feasible it would be to implement this, but submitting a well-explained feature request is likely more productive than reopening a bug that was marked as expected behavior.

daniels220, 2025-07-25:
Personally I really do think this is a bug, because:
1. It is a regression—Arachne Beta 2, waaay back when, had the desired behavior.
2. The documentation states that it should behave the way I want it to behave (thanks @brewfest!).

The simplest implementation also looks a lot more like a "bug fix" than a "new feature", in that it does not introduce any new UI or behavior, just (re-)applies existing behavior to an additional context. (Nobody has raised any possible downsides of the change that would justify making it toggleable and thus require new settings.)

I think it would be counterproductive to open a new issue, because there is a lot of good discussion on this one that would essentially just end up being copied to create a well-explained feature request as you say. I don't see why the former status of the issue matters if it is _currently_ open and there is no "expected behavior" tag applied (indeed I don't know that there ever was—certainly no comment clearly states "this is working as intended, so I'm closing the issue"). But admittedly I did say that this is borderline between a bug and an enhancement request, and yeah, I can see the argument. I'd still be more inclined to edit the description of this issue and re-tag than to create a new one, copy everything over, and still feel like I need to link back to this one for the full context, but if that's _really_ the approach the dev team would prefer, I guess I can do it.