Ultimaker/Cura #13189: [5.x] Marketplace logic sequence

Asterchades, 2022-09-01:
### Application Version

5.x

### Platform

Windows 11

### Printer

Custom FFF

### Reproduction steps

1. Open the marketplace.
2. Find an updated plugin
3. Click "Update"

### Actual results

Immediately upon clicking "Update" the "Quit Cura" dialogue will be presented. This occurs before the update has even downloaded, much less installed - and can even occur without the download beginning at all. As a result it is suggested that a user can (and probably should) immediately quit Cura to allow the new version to load, leading to a user loop where the same update will be presented on the next load and the same process can be followed again. The absence of any meaningful progress indicator makes this even more likely.

An unstable or slow connection makes this more likely but is not required.

### Expected results

The indication that you should quit Cura should be delayed until at least after the first update has actually finished installing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

While looking into this it would also be nice if the plugins available for update were sorted above the rest, rather than requiring the user to manually scan through the list trying to find them. The new UI framework has a habit of skipping entries in my experience which makes said scanning even more tedious.

MariMakes, 2022-09-02:
Hey @Asterchades,

Thanks for the report! Glad to have you here! 😄 

I was under the impression that we built functionalities to prevent this behavior.
When I install the Native CAD Plugin which is one of the biggest plug-ins there are it will show installing before prompting other screens:
[image]
But you are right that this installing doesn't pop-up if you are updating the plug-in
[image]

Good find! 👍 

If you have an update and click the gear in the marketplace, it should show you the plugins to install with the ones that require an update on top.
https://user-images.githubusercontent.com/40423138/188168554-8bb4910e-0382-4a4f-9f3a-b87804883545.mp4

I'll bring it up to the team! Fingers crossed 🤞

MariMakes, 2022-09-13:
Hey @Asterchades,

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9645

Thanks for the report! 👍