Ultimaker/Cura #13190: Cura Freeze while slicing simple shades..

tefanga, 2022-09-01:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

I Load the model... then I resize it to 20mm diameter (its a coin), after that I put the plane part on the base and then multiply the coin(with a face) 8 times...
Now I press Slice and slice for ever, it dosn't even load the bar. Im using 1.2 of LayerHeigh and 20% of refill.
[cura.log](https://github.com/Ultimaker/Cura/files/9473921/cura.log)

theres the cura log, tried to create the 3mf to upload but not case... I had to restart Cura.
The 3mf file never creates.. so I will leave the info and the cura log.
And a picture...

[image]

Greetings

### Actual results

Freeze Cura, never cotinues.

### Expected results

The slicer should have sliced and then I woulda printed :P

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I think I have add everything...

Greetings

tefanga, 2022-09-01:
Can´t upload 3mf or the file its more than 25mb...compressed the 3mf es 132MB and the original file compressed is 40MB

GregValiant, 2022-09-02:
Thanks for the report.
Can you create a 3mf file with just 1 model?  We can multiply it ourselves but we do need a model and your settings.

tefanga, 2022-09-02:
> Thanks for the report. Can you create a 3mf file with just 1 model? We can multiply it ourselves but we do need a model and your settings.

The entire model or the one Im triying to print? Because I resize the original... I can create a 3mf with only 1 resized if you want... But with 1 or 2 or 3 works ok.. but with 8 it dosnt work anymore...

tefanga, 2022-09-02:
It says file is too big.. 35mb tryed to upload... cant compress it more.. and couldn't edit the previous answer with this...

GregValiant, 2022-09-02:
_"...but with 1 or 2 or 3 works ok.. but with 8 it doesn't work anymore."_
It's a really big file.  My laptop struggles when trying to slice really big files.  I don't know if it runs out of RAM or what is going on, but it is an issue.
You could try posting the file on a remote server (like Google Drive) and provide a download link here.  Just posting the STL file might help as well.

MariMakes, 2022-09-02:
Hey @tefanga.

Welcome to the Ultimaker Cura Github! 🎉 

We really need a file for troubleshooting. 
You can also upload it to https://wetransfer.com/ and share the link so we can download it and take a look at it. 

If the file is too big it's often a sign that it's not really made of 3D printing. 
You can reduce the mesh with applications like Mesh Mixer and [instructions like this one](https://i.materialise.com/blog/en/reduce-the-file-size-of-stl-and-obj-3d-models/#:~:text=For%20example%2C%20you%20can%20give,polygon%20count%20using%20several%20options.).
That might also help with slicing it.  😄

tefanga, 2022-09-03:
Thanks for the help, the file is big indeed, including the model... but when I reduced the size thought that might help...the real size is more or less like 1 meter for example.... but I reduced to 2cm or 3cm each... even reducing it, it will cost to lag and no way to print more than 4 ?

Here is the original file: https://we.tl/t-AMMfNJdxRq

GregValiant, 2022-09-03:
At 120mb x 8 that's near a gig of models on the build plate.   Since it will slice 4 I'm thinking that your computer is running out of resources.
When you export a CAD model the resolution is set by the export utility.  That is a Hi-Res model and if it was meter with a bazillion triangles and you scaled it down then those bazillion triangles are still there, just scaled down in size.  When Cura slices it has to account for each and every one that it slices through.
I used MS 3D Builder's Simplify Tool and the file below is only 2.4mb.  Maybe it will work for you.
The model is still 1323mm in diameter.
[Resize ModeloVersaceSolidificado.zip](https://github.com/Ultimaker/Cura/files/9483493/Resize.ModeloVersaceSolidificado.zip)

This screenshot is with the resized models scaled to 2% of the original.  There are 8 and it sliced with no issues.  For size reference - the build plate is my Ender 3 Pro.
[image]

tefanga, 2022-09-03:
> 
Thanks, I will try using that app for a resize with a lot of models in the same plate... 

Greetings

MariMakes, 2022-09-05:
Yay! Thanks @GregValiant, 🎉 

The new model works for me. 
[image]

I'll be closing the isuse since the error was related to the model and not to Cura. 

It's a very cool model @tefanga, Happy Printing! 👍

closed 2022-09-05 by MariMakes