Ultimaker/Cura #13191: multiply selected doesn't multiply when print sequence is "one at a time"

atsju, 2022-09-02:
### Application Version

5.1.0

### Platform

windows 10

### Printer

Ultimaker S3

### Reproduction steps

- open an STL
- set print sequence to "one at a time"
- multiply selected => enter => no duplicate piece :(
- set print sequence to "all at once"
- multiply selected => enter => duplicated piece present :)

### Actual results

no duplication of the pieces

### Expected results

duplication of the pieces even if they do not fit the volume (manually arranging is OK)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

log of the failed multiply. Should be fairly easy to reproduce

```
2022-09-02 11:17:24,089 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]: Exception: Exception calling decoration getConvexHull: 'NoneType' object has no attribute 'getParent'
2022-09-02 11:17:24,091 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]: Traceback (most recent call last):
2022-09-02 11:17:24,092 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "UM\Scene\SceneNode.py", line 296, in callDecoration
2022-09-02 11:17:24,093 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     return getattr(decorator, function)(*args, **kwargs)
2022-09-02 11:17:24,094 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "cura\Scene\ConvexHullDecorator.py", line 114, in getConvexHull
2022-09-02 11:17:24,094 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     hull = self.getConvexHullHeadFull()
2022-09-02 11:17:24,095 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "cura\Scene\ConvexHullDecorator.py", line 131, in getConvexHullHeadFull
2022-09-02 11:17:24,096 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     return self._compute2DConvexHeadFull()
2022-09-02 11:17:24,096 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "cura\Scene\ConvexHullDecorator.py", line 325, in _compute2DConvexHeadFull
2022-09-02 11:17:24,097 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     convex_hull = self._compute2DConvexHull()
2022-09-02 11:17:24,097 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "cura\Scene\ConvexHullDecorator.py", line 306, in _compute2DConvexHull
2022-09-02 11:17:24,098 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     offset_hull = self._offsetHull(convex_hull)
2022-09-02 11:17:24,098 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:   File "cura\Scene\ConvexHullDecorator.py", line 393, in _offsetHull
2022-09-02 11:17:24,098 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]:     while ancestor.getParent() != self._root:
2022-09-02 11:17:24,099 - ERROR - [JobQueueWorker [7]] UM.Scene.SceneNode.callDecoration [298]: AttributeError: 'NoneType' object has no attribute 'getParent'
2022-09-02 11:17:24,099 - WARNING - [JobQueueWorker [7]] cura.Arranging.Nest2DArrange.findNodePlacement [53]: Object 1x2.STL cannot be arranged because it has no convex hull.
2022-09-02 11:17:24,123 - DEBUG - [JobQueueWorker [7]] UM.Operations.OperationStack.push [72]: GroupedOp.(#=0), took 0ms
2022-09-02 11:17:25,034 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [156]: New error check scheduled.
2022-09-02 11:17:25,274 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2022-09-02 11:17:25,275 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2022-09-02 11:17:25,276 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2022-09-02 11:17:25,494 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2022-09-02 11:17:25,914 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [215]: Error check finished, result = False, time = 0.9s
```

GregValiant, 2022-09-02:
Thanks for the report.
I am unable to duplicate this as Multiply works for me whether the models will fit the build plate or not.  When they don't fit then Cura should put them off to the side.  I tried with "One at a Time" enabled.
If you would create a project file (File | Save Project) and zip it and post it here it would help to debug this.

atsju, 2022-09-02:
This project as it is reproduces the problem on my computer
[UMS3_1x1.zip](https://github.com/Ultimaker/Cura/files/9478598/UMS3_1x1.zip)

GregValiant, 2022-09-02:
Thank you.
I don't know if it is something in that "Fast" profile, or a glitch in the 3mf file.  Immediately after opening the file it would not multiply the part.  Then I enabled and immediately disabled Extruder2 and everything worked correctly.

The Cura team will take a look.  When the second extruder was disabled maybe another setting didn't get the memo?