Ultimaker/Cura #13192: Tiny skin areas affect the actual print quality

Cyril-CC, 2022-09-02:
### Application Version

5.1

### Platform

Windows 10

### Printer

Snapmaker A250

### Reproduction steps

1. Import the model
2. Use the default Nomal
3. Print the model
4. result image

### Actual results

[image]

### Expected results

Cura 4.13（Wall Line Count = 2）
[image]

Cura 5.1（Wall Line Count = 2）
[image]

Cura 5.1（Wall Line Count = 3）
[image]

The skin in this location is pulling inward against the wall, is there a way to control this? 
Should this area develop skin?

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[S2A_3DP_Benchy.zip](https://github.com/Ultimaker/CuraEngine/files/9475475/S2A_3DP_Benchy.zip)

GregValiant, 2022-09-02:
Thanks for the report.
The 3mf file you attached is a model file and does not contain your printer or your Cura settings.
Use the "File | Save Project" command to create a 3mf project file and then zip the file and post the Zip here.

Cura has a lot of options and depending on how you have it set up there may be skin there or not.  You found this out when you changed from 3 walls to 2.  If you change the Line Width there will be more or less skin printed there and if you change the Layer Height it will effect the amount of skin printed there.
This is with 2 Walls and with the Line Width at .35 and Layer Height at .35.
[image]

This slice is with 3 walls at 0.2 Layer Height and 0.4 Line Width.
[image]

I've printed a lot of Benchies and never had a problem there.  If your printer has a bit of mechanical sloppiness then there can be trouble going around the bow.  The typical print order is Inner Wall, Outer Wall, (Infill), Skin.  So the outer wall should be there first and should be solid when the skin goes down.
Part of what we do as users is to adjust the Cura settings to get a good slice to make the part as we want.  I'm thinking this is a settings issue rather than a bug but I'll leave the bug label for now.

GregValiant, 2022-09-02:
This one is 2 Walls, .24 layer height, and .42 line width.  The Z seam is "Back Left" so it is at the transom corner.
[image]

The blue one is the print from those settings (PLA) and the red one is (PETG) with my normal settings of 3 Walls, .2 Layer Height and .4 Line Width.  The Wall Order is Inside to Outside.
[image]

I'm going to remove the bug label as this appears to be a settings and/or printer issue.  I'll leave it open as a discussion if you would like to comment further.  If you think it should remain open let me know.

Cyril-CC, 2022-09-05:
Thank you very much for your reply. 
Sorry to see the message today, this is a 3MF file containing parameters.
[S2A_3DP_Benchy.zip](https://github.com/Ultimaker/Cura/files/9486581/S2A_3DP_Benchy.zip)

Cyril-CC, 2022-09-05:
Personally, I think this quadrant has nothing to do with machines. I also printed this gcode on Voron 0.1 and Voron 2.4, which also have the same external wall problem.
I tried another "top layers = 2" & "bottom layers = 2", and this small area disappeared.
[image]

I have two questions to ask you.
1. Why is the "Top/Bottom Layers" default value is 5?
2. Whether to change the default value of this parameter to "line width * cos" θ "  or  " layer height * cos θ "? (θ  is the overhang angle of the layer)

Cyril-CC, 2022-09-05:
Can there be a model to automatically calculate the number of skin layers?
[image]
[image]

GregValiant, 2022-09-05:
You could put your idea into a Feature Request.
Regarding the question of _"Why is the "Top/Bottom Layers" default value is 5?"_ I don't think it is.  The "Top/Bottom Thickness" divided by the layer height gives the number of top/bottom layers.  You have 0.8 thickness in the 3mf file so .8/.16 = 5.  If you were to change the layer height to .3 you would see the number of top and bottom layers change to 3.

The gcode for this benchy is directly from your 3mf file.  I went into the gcode and changed the print and bed temps to 205 and 50 (my preferred numbers).  That was the only change I made.  This was on my Ender 3 Pro.
This result is why I'm sure the gcode is correct.  I know you have printed this on a couple of machines, but on this third machine using what is essentially your gcode I think it looks good.
[image]

Cyril-CC, 2022-09-06:
Thank you for your advice. (●’◡’●)
I will modify these two temperatures and print again.

GregValiant, 2022-09-06:
Watch your printer as it goes around the bow.  There can be no bouncing or imprecise movement.

- Belts tight.
- Trolley wheels adjusted.
- Hot end mounting screws are tight.
- Enable Accel and Jerk control at about 500 and 8 as a starting point.

Every printer is going to be somewhat different.  Mine came as a kit but I'm good mechanically and so it's calibrated as close to perfect as I can get it with the tools I have.  The temperature can vary because your thermistors are different than mine.  Heck, my bed and hot end can't agree on what room temperature is.

Cyril-CC, 2022-09-09:
Two temperatures were changed according to your parameters.
The left is the original Gcode changed, and the right is the wall printing sequence adjusted to inner wall first and then outer wall.
The picture on the left has the same problem. Is this due to insufficient heat dissipation?
[image]

GregValiant, 2022-09-09:
I think a big part of the improvement is changing to "Inside to Outside".
That problem area is at an angle that might dictate support.  As a result
when you print the outer wall first there isn't anything for that extrusion
to stick to as it is just barely over the previous layer.  When you print
Inside to Outside the inner wall is there for the outer wall to stick to
and it doesn't tend to sag.
It looks to me like you have it pretty much figured out.  That last one
looks good.

Greg

GregValiant, 2022-09-13:
I'm going to close this as it appears to be a settings or printer issue rather than Cura.  You can re-open it if you feel I'm being premature.

closed 2022-09-13 by GregValiant

Cyril-CC, 2022-09-14:
Thank you so much!