Ultimaker/Cura #13194: Unexpected error while slicing

astrokahuna, 2022-09-02:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

Error while slicing model

### Actual results

Slicing failed 

### Expected results

Slicing would have completed

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Link to project file:
https://nrchaa-my.sharepoint.com/:u:/g/personal/denniscrabtree_nrchaa_onmicrosoft_com/EXSqwHmcF2VIs30hfbiONV4BeBTbCBIb4QqPPQ1-lx2yyQ?e=qB6rer

[stderr.log](https://github.com/Ultimaker/Cura/files/9479390/stderr.log)

GregValiant, 2022-09-02:
Thanks for the report.
There are certain models that simply aren't suitable for 3D printing.  This model isn't very large but contains 5.3million vertices and 1.8million triangles.  That is a fairly ridiculous density/resolution.  The guard rails that are on some of the platforms are especially troublesome to print.

The model is watertight but has a lot of other errors (duplicate faces, degenerate faces, etc.).  I couldn't get MeshMixer to work with it.  Cura never finished slicing with your UM S3 printer active.  I was able to repair the file with MS 3D Builder.

I configured my Ender as a dual extruder printer and after moving the model away from the build plate midpoint and rotating the model (that is a known bug) I did get the repaired model to slice at a 0.2 Layer Height.
[image]

So there are multiple things going on here.  The model itself is a PITA to work with as well as having errors, but I think there may be something going on with the Ultimaker S3 and your Normal profile as well.  In addition there is the "won't slice when centered" bug which has come up numerous times with Cura 5.1.

Someone from the Cura Team (@MariMakes ?) will take a look at this.

astrokahuna, 2022-09-06:
> ***Attention*** This email originated from outside of the NRC. ***Attention*** Ce courriel provient de l'extérieur du CNRC.

Thanks for the response Greg. Moving the model off-center solved the issue.

Cheers
Dennis

----------------------------------------------------------------------------------------------------
Dr. Dennis R. Crabtree
Director Emeritus, Dominion Astrophysical Observatory / Directeur émérite Observatoire federal d’astrophysique - W̱SÁNEĆ and Lkwungen traditional territory
Herzberg Astronomy & Astrophysics / Herzberg astronomie & astrophysique
National Research Council / Conseil national de recherches Canada

[image]<https://user-images.githubusercontent.com/64202104/188216969-fb0654b5-2827-4503-baec-c4d3396886fd.png>

Someone from the Cura Team ***@***.***<https://github.com/MariMakes> ?) will take a look at this.