Ultimaker/Cura #13198: Tooltip get in the way of the Start/End G-code setting of printer

edolis, 2022-09-03:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

any

### Reproduction steps

1. open Machine settings
2. try editing G-code in the textbox

### Actual results

tooltip popping up continuously  anywhere in the text box

### Expected results

no tooltip at all. Bind the tooltip to the label, pretty please.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Just a pretty annoying thing.
When editing start or end G code in Printer Settings, the tooltip continuously gets in the way and modifying the code turns into an annoyance, Please fix this
[image]

MariMakes, 2022-09-07:
Hey @edolis ,

Thanks for the report 👍 
 
I don't work a lot with the Start/End Gcode screen but I can imagine this tooltip can become quite annoying if you work with it a lot.  I'll bring it up with the team to see if we can do something about it to make it less annoying.

Fingers crossed 🤞

MariMakes, 2022-09-13:
Hey @edolis,

I brought it up to the team and we've added it to the backlog with the intent to make the tooltip easier to work within the start and end gcode. For internal reference CURA-9643

Fingers crossed! 🤞

Drewbacca1981, 2022-10-02:
It's even worse when you have the Settings Guide plugin.  It takes up the entire text box, and makes editing the start and end g-code boxes all but impossible.

WebSpectrum, 2023-03-05:
This still seems to eb a problem - very annoying

GregValiant, 2026-08-19:
Along with the Varialble Line Width, Cur 5.0 changed the software used for the dialogs and controls.  It was not an easy transition.
I'll close this as now (5.13.0) it is not longer an issue.

closed 2026-08-19 by GregValiant