Ultimaker/Cura #13200: Slicer stops

ETS5, 2022-09-03:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality Pro V2

### Reproduction steps

I loaded my design and pressed slice and it failed
tried again after a reboot and failed again

### Actual results

Just errors out after it tries to load for a few seconds

### Expected results

slice my file to gcode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Please let me know what you need from me I can make this happen over and over again

GregValiant, 2022-09-03:
Thanks for the report.
There have been a lot of reports of this behavior in Win10.  You may be able to work around it by moving the model away from the midpoint o the build plate and/or rotatiing the model.
It would be good to have a project file.  The more models that can be checked the greater the chance this will be fixed.
Load the model and use the "File | Save Project" command.  Zip the 3mf file and post the zip file here.

MariMakes, 2022-09-07:
Hey @ETS5,

Welcome to the Ultimaker Cura Github! 🚀 
@GregValiant is right, we'll need a project file to help you troubleshoot and better understand if you are encountering a problem that is similar to our other report.

Can you share your project file with us? Thank you 👼

ETS5, 2022-09-09:
[FlyingSerpent_2-26_Curled.zip](https://github.com/Ultimaker/Cura/files/9539010/FlyingSerpent_2-26_Curled.zip)
Here is the file that I was trying to slice I did not try moving it around or anything like that. I am also now on a Win11 machine and still have the same issue

GregValiant, 2022-09-10:
AAhhh, the infamous "Crystal Dragon" but this one has feathers.  There are a LOT of errors in that model.
This is the view in MeshMixer.  All the red and magenta balls indicate errors.  The entire tail is a mess.  Mesh Mixer was not able to repair the model.
[image]

I'm going to remove the bug label.  That model cannot be made watertight and even if you get it to slice it probably won't print correctly.
I'll leave this ticket open if you would like to respond.

ETS5, 2022-09-10:
[FlyingSerpent_3-2_BALLWINGS_Curled4.zip](https://github.com/Ultimaker/Cura/files/9540658/FlyingSerpent_3-2_BALLWINGS_Curled4.zip)
What is funny is that this model slices no issue just the wings are different.

Thank you for sharing that with me, I will see about correcting the issue in my cad

GregValiant, 2022-09-10:
It depends on what type of errors are in the model.  Cura can work around some types while others just bring a slice to a stop.
Here is that second model in MeshMixer.  Once again you can see that the tail is a mess.
[image]

ETS5, 2022-09-11:
ok well thank you for your time and sorry for opening up a bug ticket and it just being my model as a mess.

closed 2022-09-11 by ETS5