Ultimaker/Cura #13203: Support x/y distance/override not respected.

PhilBaz, 2022-09-04:
### Application Version

5.0

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

use & change support x/y distance on print with support.

### Actual results

X/y distance doesn't change

### Expected results

changes..

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[support xy.zip](https://github.com/Ultimaker/Cura/files/9483697/support.xy.zip)

Asterchades, 2022-09-04:
Try updating to 5.1. I'm unable to reproduce your issue with the provided project and profile, meaning if there was an issue with 5.0 it has since been corrected - and updating would be the only way you would get any fixes anyway.

[image]
2.0mm

[image]
0.4mm

closed 2022-09-04 by Asterchades

PhilBaz, 2022-09-04:
@Asterchades 

But what about the initial layers? They would fuse to the print. they have not moved.

PhilBaz, 2022-09-04:
@Asterchades 
Oh, no...im sorry. you are right. bad model.