Ultimaker/Cura #13205: issues

rhender25, 2022-09-04:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Lotmaxx Shark

### Reproduction steps

Ever since I upgraded to 5.1.0 EVERYTIME I try to change certain digits (such as infill density) It wont let me put the digits in except 1 at a time ,such as 40 % I put in 4 and  it kicks me out and I have to go back in to put the zero I never have this problems on other cura programs just this one I've tried to install and reinstall , upgrade but still the same it gets pretty annoying 

### Actual results

bug

### Expected results

bug

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

bug

GregValiant, 2022-09-05:
Thanks for the report.
I haven't heard of this behavior before and it doesn't occur on my Win10 system.
Can you provide more information about your computer system (processor, video system, RAM, etc.)?  Someone from the Cura Team will take a look.

rhender25, 2022-09-05:
Device name BIG27Processor Intel(R) Core(TM) i7-4770T CPU @ 2.50GHz
2.50 GHzInstalled RAM 12.0 GB (11.9 GB usable)Device ID
1D86EA3A-15F4-4873-A09B-8F5FC77CF7B7Product ID
00331-20442-00000-AA208System type 64-bit operating system, x64-based
processorPen and touch Touch support with 10 touch points Edition
Windows 10 ProVersion 21H2Installed on ‎12/‎9/‎2021OS build
19044.1889Experience Windows Feature Experience Pack 120.2212.4180.0
Intel(R) HD Graphics 4600NVIDIA GeForce 730A

	-----------------------------------------From: "GregValiant" 
To: "Ultimaker/Cura"
Cc: "randy25", "Author"
Sent: Monday September 5 2022 12:14:46PM
Subject: Re: [Ultimaker/Cura] issues (Issue #13205)

	Thanks for the report.
 I haven't heard of this behavior before and it doesn't occur on my
Win10 system.
 Can you provide more information about your computer system
(processor, video system, RAM, etc.)? Someone from the Cura Team will
take a look. 

	—
[2].

Links:
------
[1]
https://github.com/Ultimaker/Cura/issues/13205#issuecomment-1237266427
[2]