Ultimaker/Cura #13206: Weird Printing

bcw219, 2022-09-04:
### Application Version

5.1 beta

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Used Paint 3D to create a cone shape with a couple doughnut shapes around it to create the shape of a plug with ridges.
2. Loaded the 3MF file into Ultimaker Cura, putting the point downwards.
3. Placed a cube shape around the plug shape.
4. Used the Per Model settings to create a cutting mesh of the plug shape with zero walls and zero infill.
5. Slice the print which should only take an hour to print. The plug is only 25mm by 25mm by 30mm, while the cube is 30mm by 30mm by 30mm.
[image]

### Actual results

After leveling, it starts to make the movement to clear the nozzle of excess material. It goes to the back of the plate perfectly normal, but when it tries to return to the origin, it rapidly speeds up. It will continue to print as if it hadn't sped up randomly. It gets maybe two layers in and stops moving. The printer stays heated, provides no message, and can only be turned off with the power switch. The screen is no longer usable.

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_plug.zip](https://github.com/Ultimaker/Cura/files/9485267/CE3PRO_plug.zip)

GregValiant, 2022-09-05:
Thanks for the report.
I added a 30mm cube as the model and it sliced fine.  The gcode is fine as well.
With my setup I can't reproduce the problem.  FYI my printer is also an Ender 3 Pro and the gcode would print fine on it.

Post a gcode file that you are having trouble with.  If you can - take it off the SD card and zip and post that file.  If there is a bad sector on the memory card that gcode file may be corrupt.

bcw219, 2022-09-05:
[CE3PRO_plug.zip](https://github.com/Ultimaker/Cura/files/9491371/CE3PRO_plug.zip)
I attached the file directly from my SD card after I have tried modifying it in Ultimaker. It has multiple plug shapes this time, but it still stopped partially through the second layer.

bcw219, 2022-09-05:
I formatted the SD card, and it seems to have fixed the problem. It's strange because it was the only one I had a problem printing. Any of the previous prints would print, but not that one.

GregValiant, 2022-09-05:
It can depend where the bad sectors are on the card.  Not every gcode file on there may have been written across bad sectors. 
 Good that you figured it out.

That gcode file you posted is 0kb.  It's an empty file.
I'll close this as it appears to be a bad memory card problem.  If you think that was premature let me know.

closed 2022-09-05 by GregValiant