Ultimaker/Cura #13207: Error Slicing

DevGuyRob, 2022-09-04:
### Application Version

5.0.1

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Rotate stl file (attached stl, and screenshot). Sliced on Visual Intent .15 with Supports (20%infill). Would upload file but can't since it is a .stl

[image]. 
[1712-14-_14_bodyPart_g_V2_FDM (repaired).zip](https://github.com/Ultimaker/Cura/files/9485278/1712-14-_14_bodyPart_g_V2_FDM.repaired.zip)

[UMS3_1712-14-_14_bodyPart_g_V2_FDM (repaired).zip](https://github.com/Ultimaker/Cura/files/9485285/UMS3_1712-14-_14_bodyPart_g_V2_FDM.repaired.zip)

### Actual results

Generic Slicing failed Error

### Expected results

Sliced :)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Thanks.... included project file, original stl, screenshoot

GregValiant, 2022-09-05:
Thanks for the report.
With an S3 active and Cura set up to print with Extruder 1, with the model at the midpoint of the build plate it would not slice for me.  Moving the model to the right rear corner allowed the model to slice.
This bug has been reported several times and I believe is being repaired for 5.2.  A workaround is to move and/or rotate the model on the Cura build plate.
I'll mark this as a duplicate.  (FYI this problem has so far only occurred on Windows systems.)

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker