Ultimaker/Cura #13216: Cura 5.1.0 unable to slice simple .stl.

holyloktar, 2022-09-05:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Maker Select V2

### Reproduction steps

I opened the attached .stl, shrunk it by 50%, put it flat on the bed and tried to slice it and it failed. I can slice the original file and I can shrink and slice it in 5.0.0 but 5.1.0 fails.
[moyahidev1.zip](https://github.com/Ultimaker/Cura/files/9492257/moyahidev1.zip)

### Actual results

Cura just says slicing failed.

### Expected results

You will get the error attached.
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9492266/cura.log)
[WDI3_moyahidev1_Project.zip](https://github.com/Ultimaker/Cura/files/9492268/WDI3_moyahidev1_Project.zip)

GregValiant, 2022-09-06:
Thanks for the report.
After scaling it to 50% and moving it around a bit a bit on the build plate I did get it to slice.  This behavior has been noted in other bug reports so I'll make this a duplicate.
This is the 50% preview.  You can see that a portion of the model is too thin to print with the .4 nozzle and line width that I used.  The same area is a problem at 100% scale.  Although the file doesn't appear to have errors - it isn't really suitable for FDM printing unless that area is made thicker.
[image]

holyloktar, 2022-11-01:
Thanks for the response, and sorry for the long delay. I appreciate you looking at it.

closed 2022-11-01 by holyloktar