Ultimaker/Cura #13218: View, center of rotation

porzioev, 2022-09-06:
### Is your feature request related to a problem?

1. at any time
2. trying to rotate the view
3. since the pivot point is too far from the center of the scene, the objects and the printing plate splash out of the frame.

### Describe the solution you'd like

solution:
Set the pivot point in the center of the depth of field or object or simply in the center of the build plate. 

### Describe alternatives you've considered

Maybe making the center of rotation selectable from the settings. Or rather by making the center of rotation visible and selectable as in photoshop or cad software.

### Affected users and/or printers

The usability of the program itself.

MariMakes, 2022-11-01:
Hey @porzioev 

Sorry, it took us a while to get back to you 😞 

Are you aware of the preference Center Camera when Item is Selected?
This seems to be a solution to what you are describing. 
[image]

Would you still like to keep this issue open or do you agree that we can close it?

porzioev, 2022-11-02:
Yes this is a good solution. Also if it would be better to dynamically select the pivot point just by auto detecting from mouse position at the moment of click and drag.