Ultimaker/Cura #13220: failed to run cura from source

ilovelib, 2022-09-06:
### Application Version

5.0

### Platform

windows 10

### Printer

s5

### Reproduction steps

follow the web:

https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#3-initialize-the-virtual-python-development-environment

### Actual results

ERROR: pyarcus/5.2.0-alpha+testing_29@ultimaker/testing: Error in requirements() method, line 49
        for req in self._um_data()["requirements"]:
        KeyError: 'requirements'

**Here is the whole output info** 
D:\cura_dev\Cura>conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

conanfile.py (cura/None): Using conandata.yml defaulting to testing channel
pyarcus/5.2.0-alpha+testing_29@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: .conan while recipe is build for version: 5.2.0-alpha+testing_29
ERROR: pyarcus/5.2.0-alpha+testing_29@ultimaker/testing: Error in requirements() method, line 49
        for req in self._um_data()["requirements"]:
        KeyError: 'requirements'

### Expected results

no error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

jellespijker, 2022-09-08:
Can you try again, I have updated the recipes yesterday. make sure you use the latest configuration;
```
conan config install https://github.com/ultimaker/conan-config.git
```

jellespijker, 2022-09-08:
I see that you're using Windows, be aware that we still have on open issue with setting up the virtual environment for Windows from source see issue: https://github.com/Ultimaker/Cura/issues/12773

I'm currently working on this issue

ilovelib, 2022-09-08:
> I see that you're using Windows, be aware that we still have on open issue with setting up the virtual environment for Windows from source see issue: #12773
> 
> I'm currently working on this issue

Hi, Jellespijker, I reran the commands from 1 to 3, unfortunately same issue.

pynest2d/5.2.0-alpha+testing_30@ultimaker/testing:
pynest2d/5.2.0-alpha+testing_30@ultimaker/testing: ERROR: Package '98633b7ddf8934c5fbc2d9c9264b852c3e7ff0b9' build failed
pynest2d/5.2.0-alpha+testing_30@ultimaker/testing: WARN: Build folder C:\.conan\008eb0\1\build\Release
ERROR: pynest2d/5.2.0-alpha+testing_30@ultimaker/testing: Error in build() method, line 100
        cmake.build()
        ConanException: Error 1 while executing cmake --build "C:\.conan\008eb0\1\build\Release" -- -j8
PS D:\cura_5.1.0\Cura>

jellespijker, 2022-10-10:
Can you try again using the instructions in this comment

https://github.com/Ultimaker/Cura/issues/12773#issuecomment-1272992115

closed 2022-10-25 by github-actions[bot]