Ultimaker/Cura #13221: [Cura 5.1.0] Rotate item to place a selected face on Bed don't work

zebons, 2022-09-06:
### Application Version

5.1.0

### Platform

Windows 10 & 11

### Printer

Voron 2.4 350

### Reproduction steps

1- Load an item and select it
2- Choose rotation tool and select the thrird tool 'select the face to bed'
3- then select the face that you'd be directly onto the bed plte

### Actual results

The item rotate but not as asked and the selected face isn't placed on bed plate. And it's very difficult to select the face that we want to be onto the bed

### Expected results

As it was working fine in 5.0.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no need to add something else. Follow instruction given above

zebons, 2022-09-07:
Strange I redo a test this morning and it's working fine with the same item thet won't working few days ago... So I don't know  if this issue must be close or not

MariMakes, 2022-09-08:
Hey @zebons,

Thanks for the report. 👍 
I did a few experiments and did not see the behavior in Cura 5.1.

Do you have a project file for us? 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files you may need to use WeTransfer or similar file sharing sites.

Also, about your comment. You are right. If we cannot reproduce the incorrect behavior we cannot see if the implemented fix works. If we cannot reproduce it we will have to defer the ticket until we have solid steps to reproduce it.

zebons, 2022-09-08:
Hi, As said before I cann't reproduce the issue now even with the item 
for which it wasn't working. So I think that the issue must be close for 
now.
If I encounter again the issue I'll re-open a new one with more 
investigation.

Thanks for all

closed 2022-09-08 by zebons