Ultimaker/Cura #13225: It won't let me open files on the AppImage 64 Bit

GhostPisces, 2022-09-06:
### Application Version

5.1.0

### Platform

Linux App Image

### Printer

Neptune 2

### Reproduction steps

I went to open my files and it wouldn't open them. I deleted the file and redownloaded it and that still didn't work.

### Actual results

I still cant open my files to slice them in the program.can't

### Expected results

I should be able to open my files and slice them to print them.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

It just won't let me open files for slicing.

notfood, 2022-09-08:
I have the same issue.

Tested the following:
- `PASS` Ultimaker_Cura-4.13.1.AppImage
- `FAIL` Ultimaker-Cura-5.0.0-linux.AppImage
- `FAIL` Ultimaker-Cura-5.1.0-linux.AppImage
- `FAIL` Ultimaker-Cura-5.1.0-linux-modern.AppImage 

Anything above 5.0 fails to open the file browser. You can open files by drag and drop, but then you can't export the gcode because it doesn't open either.

OS: ArchLinux
Kernel: 5.19.3
DE: KDE Plasma Desktop 5.25.5

notfood, 2022-09-08:
After further investigation, I got it to show the file browser panels by prepending `LD_PRELOAD=/usr/lib/libstdc++.so.6` to the command.

Information here: https://aur.archlinux.org/packages/cura-appimage-bin

MariMakes, 2022-09-13:
Hey @notfood,

Thanks for the report 👍 
Sorry, it took us a while to get back to you. 😞 
I'm happy to see you were able to do some troubleshooting on your own 💪 

Can I interpret your last comment as your issue is resolved?
Or do you need someone from the team to look at your issue? 🤔

notfood, 2022-09-14:
For me it has a workaround, probably other's will have the issue in the future.

closed 2022-09-27 by github-actions[bot]