Ultimaker/Cura #13226: Unexpected error when slicing

jelalanne, 2022-09-06:
### Application Version

5

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Asked to slice an object, that's it. 

### Actual results

Crashes, cannot slice, unexpected error pops up.

### Expected results

Slice correctly

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_VadorsBladeUnstableSleeve_side v11.zip](https://github.com/Ultimaker/Cura/files/9498503/CE3PRO_VadorsBladeUnstableSleeve_side.v11.zip)
[stderr.log](https://github.com/Ultimaker/Cura/files/9498499/stderr.log)

jelalanne, 2022-09-06:
_Update_:
Works on Low Quality only

jelalanne, 2022-09-06:
_Update 2_:
Worked when I lowered Initial Layer Height just a bit, it was at 0.2mm like Layer Height, owered it to 0.15 and it sliced.

GregValiant, 2022-09-06:
Thanks for the report.
This is a duplicate of many other reports.  The problem appears to be with models that are at 0,0 on the build plate.  Moving the model and rotating it 90° often works to get the model to slice.
I believe this bug is being worked on and will be fixed for Cura 5.2.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker