Ultimaker/Cura #13228: Error when slicing

gacm89, 2022-09-06:
### Application Version

5,1

### Platform

windows 11

### Printer

BIQU B1

### Reproduction steps

after a change in the parameters "skin expand distance" the slicer was unable to slice.

### Actual results

I change it back to the previous value, no change. 
Then I change the layer height and it stopped the error.

### Expected results

Stop the error.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

no comments

GregValiant, 2022-09-06:
Thanks for the report.
Without a project file or log we can't do an investigation.
The Cura.log file can be found by using the "Help / Show Configuration Folder" command.
To make a Project File - load you model in Cura and then set up to slice.  Use the "File | Save Project" command to create a 3mf Project File.
Put the log file and the 3mf file into a zip folder and post the Zip here.

gacm89, 2022-09-06:
Hi Greg
Follow the log file
[curaLOG.zip](https://github.com/Ultimaker/Cura/files/9501418/curaLOG.zip)

I notice that the bug happens when I try to slice in 0,5mm layer thickness.
Follow the project file as well.
[BB1_seletor_330ml_REV02_sup.zip](https://github.com/Ultimaker/Cura/files/9501412/BB1_seletor_330ml_REV02_sup.zip)

GregValiant, 2022-09-07:
I looked at the project file.
This one is a head scratcher.  First off, I checked the model and it's watertight and doesn't appear to have any errors.

The model will not slice when Horizontal Expansion AND Horizontal Hole Expansion are equal and <> 0 OR the Initial Layer Height is less than the Layer Height.
If either of those conditions change then the model will slice at a Layer Height of 0.5.
If you move the model on the build plate and rotate it 90° then the model will slice with your current settings and my above statement becomes irrelevant.
So there are a couple of workarounds available to you:  Change the Initial Layer Height to 0.5 or change the Horizontal Expansion to 0.13.  You could also move and rotate the model.
I believe the Cura Team is working on this bug and it should be fixed for 5.2.
[image]

gacm89, 2022-09-11:
Hi Greg, thanks for the hints.
After it, I change the layer height to 0,3mm and I sliced the model (yeah... I don't understand as well).
Today, I used the same working parameters for another set of parts and.... error....
As you said, I hope Cure Team can fix it in version 5.2.
Remark: I've never had slicing errors before version 5.1.

GregValiant, 2022-09-11:
I threw that description in there for the Cura team.  It might give them a better chance of chasing this down.
This bug comes up a lot so yes, let's hope they fix it.