Ultimaker/Cura #13231: Slicing Failed

Angus79, 2022-09-06:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Attempted to slice a commercially made stl file using a pre-made slicing profile.

### Actual results

Received an error message that slicing failed.
The same stl file slices fine in Creality slicer 4.8.2 using a self-made prfile.

### Expected results

Slicing the file should have  produced a usable .gcode file for printing.

[CE3PRO_851-28-28_platform_4_FDM(repaired).zip](https://github.com/Ultimaker/Cura/files/9500711/CE3PRO_851-28-28_platform_4_FDM.repaired.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-09-06:
Thanks for the report and the project file.
It sliced fine for me on the first try.
There have been a lot of reports lately regarding a bug when slicing when running Windows.  A workaround is to move the model on the build plate and then rotate the model.  Sometimes that is able to overcome the bug.
I'll leave this as a bug report and mark it as a duplicate.
This problem should be fixed in 5.2.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker