Ultimaker/Cura #13234: No walls showing in preview

Champ1800, 2022-09-07:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Open any .stl file & slice it with my standard profiles.

### Actual results

In the preview window none of the walls or outer layers are showing, only the infill.
The attached project file is one I have used before without any issue.

### Expected results

Normal preview to examine the sliced item layer by layer

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_GizoSpider.zip](https://github.com/Ultimaker/Cura/files/9504130/CE3_GizoSpider.zip)
[image]
[image]
[image]

GregValiant, 2022-09-07:
Thanks for the report.
In the preview screen you can see a DropDown list for "Line Type".  Click on it and make sure you have Shell and Helpers turned on.
Let us know if turning them on makes the outer and inner walls visible.  When I slice it I get this...
[image]

Champ1800, 2022-09-08:
That fixed it, thank you. Sorry it was a stupid issue on my side. I must have accidently unchecked it.

closed 2022-09-08 by Champ1800