Ultimaker/Cura #13236: Support Structures still calculated for portions of objects removed by cutting objects.

rjsimon, 2022-09-07:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality CR-10s

### Reproduction steps

1) I imported an object for print.

2) I created two Support Blocks and placed them over parts of the model's geometry that I did not want to print.

3) I set the mesh type of my support blocks to "Cutting Object" via the drop-down and set Wall Line Count, Top Layers, Bottom Layers and Infill Density to '0'.

4) I enabled supports and set the Support Type to 'Tree', and Support Placement to 'Touching 'Buildplate'.

4) I then hit the slice button.

### Actual results

The Slicer correctly removes tool paths from the volumes of the cutting Mesh where the model once was but plots the support structures for the phantom parts nonetheless adding time and material to the print.

[image]

[image]

### Expected results

The slicer should not add support structures where the volume of a Cutting Mesh excludes that geometry from printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Cutting Mesh Support Structure Bug Example.zip](https://github.com/Ultimaker/Cura/files/9504514/Cutting.Mesh.Support.Structure.Bug.Example.zip)

GregValiant, 2022-09-07:
Thanks for the report.
You must add the "Generate Support" setting to your Cutting Meshes and then turn it off.
[image]

[image]

I'll remove the bug tag as this is a setting issue.  Let me know if you feel it is premature.

rjsimon, 2022-09-07:
No, absolutely feel free to close the ticket. Thanks for the speedy response and effective solution! 
Thanks also for maintaining this piece of software. It is incredibly impressive.

closed 2022-09-07 by rjsimon

rjsimon, 2022-09-08:
[Cutting Mesh Support Structure Bug Example.zip](https://github.com/Ultimaker/Cura/files/9522503/Cutting.Mesh.Support.Structure.Bug.Example.zip)

Attached here is a file that illustrates the issue. Grouping these support meshes causes them to drop as described.
Please disregard the file name.

GregValiant, 2022-09-08:
Thank you.
The Cura team will take a look at this.
What I see is that the first time the two support blockers are grouped they change location within their bounding box.  The origin point of the group does not change or move.
Ungrouping and then re-grouping, they act as would be expected.