Ultimaker/Cura #13240: Slicing Failure

sparker0594, 2022-09-07:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Loaded my file,
Sliced the file using my normal setting (custom)

File Link: https://www.dropbox.com/sh/jsoyd5dqkb6pirv/AABL7TNvLZYnNsNaSxeL5akQa?dl=0

### Actual results

The slicing of the file failed with the progress bar approx 60% through.
Slicing failed message box appeared

### Expected results

completed slicing and g-code produced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

File Link: https://www.dropbox.com/sh/jsoyd5dqkb6pirv/AABL7TNvLZYnNsNaSxeL5akQa?dl=0

GregValiant, 2022-09-07:
Thanks for the report.
This is a duplicate report for this bug.  A workaround is to move the model on the build plate and rotate it 90°.  It should slice.
This bug should be fixed in 5.2

sparker0594, 2022-09-07:
Thanks for the feedback and instruction!

Very quick response!