Ultimaker/Cura #13243: Tree supports placement make no sense

Behem07h, 2022-09-08:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender-3/Ender-3Pro/Ender-3-3S

### Reproduction steps

1. Try to slice a model with many overhanging areas in need of supports
2. Inspect the support placement

### Actual results

1. Observe that suports do not stretch as far horizontally as they should be, based on the configured branch angle
2. Observe that fresh tree supports get placed ontop of tree supports, rather than extending the lower tree upwards
3. Also observe that there are floating supports and areas of the print that are entirely unsupported despite having abundant space for supports and an obvious need for them

### Expected results

1. Tree supports should not have clearly erroneous placement

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[lord_relictor_icon E3V2 W.15 H.08 High.3mf.zip](https://github.com/Ultimaker/Cura/files/9527174/lord_relictor_icon.E3V2.W.15.H.08.High.3mf.zip)

GregValiant, 2022-09-08:
Thanks for the report.
Did you change anything between taking your screenshot and creating the Project File?  The slice I did from your project is a lot different.
[image]

Looking at your settings - how confident are you that you can print that with a 0.4 nozzle at 0.15 Line Width?  The tips at the top and this area show some discontinuities.  Pieces would fall off.
This doesn't look so much like a bug as it does a model that needs some adjustment to be able to print it vertically.
[image]

Behem07h, 2022-09-08:
There was no change of settings between me slicing the model and generating
the project file that i sent. I've noticed this in some other reports as
well - is this hinting at a hardware specific bug?

I'm actually using a .15 nozzle, my printer profile is set to 0.4 because
thats the widest nozzle i use on this printer. I've already manually
overridden the settings that use nozzle diameter to calculate rather than
line width, so i believe there should be no issues with that?

GregValiant, 2022-09-09:
I think it's a question of how the Setting Profile, Nozzle, and Quality files are getting along.  I had installed a .15 nozzle in Cura for my Ender3P but I get the "There are no profiles matching the configuration of this extruder" warning.
I had never noticed that before.  I'll play with it this afternoon and try to get it to work.

MariMakes, 2022-09-15:
Hey @Behem07h,

Thanks for the report 👍 

I'm running into the same issue as @GregValiant.
If I slice the file that you shared, my supports make way more sense. 
[image]

Could it be that you reduced the Support Z Distance or increased Support X/Y Distance?
Because then it might be a duplicate of:
https://github.com/Ultimaker/Cura/issues/8614

Can you let us know if this is a duplicate or provide us with a file with which we can reproduce your issue? 

Thank you! 😇

DerGenaue, 2022-09-25:
Tree supports in their current implementation are not deterministic and some details are hardware-dependent (# of threads) which can cause issues like this together with the standard implementation's behavior of dropping branches if it doesn't know what to do with them.

You can try out this [alternative tree support implementation](https://github.com/ThomasRahm/CuraEngine) if you want, that one might solve your problems in the meantime.

closed 2022-09-30 by github-actions[bot]

DerGenaue, 2023-07-04:
FYI: This probably has been solved in the new Cura 5.4 release