Ultimaker/Cura #13245: Brim not being printed in a spiral from the outside inwards (v5.1.0)

dhysuiej, 2022-09-08:
### Application Version

5.1.0

### Platform

MacOS

### Printer

Ender 3 v2

### Reproduction steps

1. Slice the attached STL file in Cura 5.1.0.
2. Visualize the first layer.

### Actual results

The brim is not printed in a spiral from the outside inwards. Instead the contour is printed first and afterwards the remaining brim is printed from inside outwards. The result is that one can't make manual adjustments to the bed level before the brim reaches the part where it touches the actual piece that is printing.

[image]

### Expected results

The brim should be printed from the outside inwards to give enough time to make manual bed level adjustments before the actual piece starts to print.

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Archive.zip](https://github.com/Ultimaker/Cura/files/9528447/Archive.zip)

GregValiant, 2022-09-08:
Thanks for the report.
This may be a duplicate of #11936.  I don't use brims often but I have noted the behavior on skirts as well.  For the skirts, it doesn't happen all the time so I think I may be hitting a combination of settings that is triggering the behavior.

dhysuiej, 2022-09-08:
Thanks. Closing this issue then.

closed 2022-09-08 by dhysuiej