Ultimaker/Cura #13247: Feature: per-material flow setting

andrewmeyer, 2022-09-08:
### Is your feature request related to a problem?

since flow rates tend to vary per material, it would be very helpful to store flow as a setting for each material.

### Describe the solution you'd like

The Material print settings should be expanded to include:
- flow

Optionally include initial layer flow.

All other flow settings should key off of the base flow as the currently do

### Describe alternatives you've considered

currently I write the flow percentage on my spool and record it on my swatches.

### Affected users and/or printers

All

Asterchades, 2022-09-09:
Try the Material Settings plugin, available for free on the Marketplace. It allows you to put any Cura setting you like - even those added by other plugins, such as the Z Offset and Linear Advance plugins - into the Material Properties list, letting you configure them on a per-material basis.

It would also be possible to make additional print profiles which contain the altered setting, but I can understand not wanting to do that as it will prevent other setting changes from propagating across them (unless you manually edit your own printer definition).

andrewmeyer, 2022-09-12:
Thanks for the tip @Asterchades, this does indeed add the setting.  I still feel like this should be rolled into the core application instead of held in an addon since every filament will likely have a different flow.  That said I also understand the interest in keeping the UI as simple and clean as possible for less picky printers.

This is probably simply a matter of adding flow to the array of fields stored by a material from a coding perspective.  If it does get implemented I will be more than happy to test a development build, my python is just non existent.  Too bad its not perl! :laughing: