Ultimaker/Cura #13248: Unable to open file manager linux appimage

Aaron7317, 2022-09-08:
### Application Version

5.1.0 modern

### Platform

Arch Linux

### Printer

Creality CR-10S

### Reproduction steps

I run the cura modern AppImage and everything seems to work fine on Sway/Wayland. However, whenever I click on something that would usually open up a file manager nothing happens (Importing a profile, opening a file, creating a new project, etc). I have tried installing several different file managers thinking that was the issue (vifm, pantheon-files, nautilus, dolphin) but none of them were opened by Cura.

### Actual results

Cura is unable to open any files of any sort because no file manager is opened when trying to access files. Example: Click on "open file" and nothing happens. Click on "New project" and nothing happens. Thanks for taking the time to review this.

### Expected results

A file manager of some time should be opened in a new window in order to select the file I want to access.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

2022-09-08 16:58:56,571 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [548]: Loading metadata into container registry took 0.574455738067627 seconds
2022-09-08 16:58:56,582 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2022-09-08 16:58:56,584 - INFO - [MainThread] cura.CuraApplication.run [823]: Initializing machine error checker
2022-09-08 16:58:56,765 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition creality_cr10s failed to pickle.
2022-09-08 16:58:56,766 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [creality_cr10s]
2022-09-08 16:58:56,885 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition creality_base failed to pickle.
2022-09-08 16:58:56,914 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [192]: Extruder[creality_base_extruder_0 #2] added to [Creality CR-10S] at position [0]
2022-09-08 16:58:56,926 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Creality CR-10S in list of firmware to check.
2022-09-08 16:58:57,134 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_cr10s took 0.20318150520324707 seconds.
2022-09-08 16:58:57,140 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ChangeAtZ
2022-09-08 16:58:57,141 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ColorMix
2022-09-08 16:58:57,142 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: CreateThumbnail
2022-09-08 16:58:57,143 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2022-09-08 16:58:57,144 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: DisplayProgressOnLCD
2022-09-08 16:58:57,144 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: FilamentChange
2022-09-08 16:58:57,145 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: InsertAtLayerChange
2022-09-08 16:58:57,146 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: PauseAtHeight
2022-09-08 16:58:57,147 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: RetractContinue
2022-09-08 16:58:57,147 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: SearchAndReplace
2022-09-08 16:58:57,150 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: Stretch
2022-09-08 16:58:57,151 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: TimeLapse
2022-09-08 16:58:57,151 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: UsePreviousProbeMeasurements
2022-09-08 16:58:57,152 - WARNING - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadAllScripts [155]: Unable to create a folder for scripts: /tmp/.mount_curaw5CpbF/share/uranium/resources/scripts
2022-09-08 16:58:57,152 - WARNING - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadAllScripts [155]: Unable to create a folder for scripts: /tmp/.mount_curaw5CpbF/share/cura/resources/scripts
2022-09-08 16:58:57,170 - INFO - [MainThread] cura.CuraApplication.run [828]: Initializing machine manager
2022-09-08 16:58:57,174 - INFO - [MainThread] cura.CuraApplication.run [833]: Initializing container manager
2022-09-08 16:58:57,283 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition fdmprinter failed to pickle.
2022-09-08 16:58:57,285 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.loadAuthDataFromPreferences [258]: Attempting to load the auth data from preferences.
2022-09-08 16:58:57,289 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [156]: New error check scheduled.
2022-09-08 16:58:57,290 - WARNING - [MainThread] UM.OutputDevice.OutputDeviceManager.removeOutputDevice [235]: Could not find output device with id digital_factory to remove
2022-09-08 16:58:57,312 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2022-09-08 16:58:57,503 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [215]: Error check finished, result = False, time = 0.2s
2022-09-08 16:58:57,509 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [188]: Loading theme file: /tmp/.mount_curaw5CpbF/share/cura/resources/themes/cura-dark/theme.json
2022-09-08 16:58:57,509 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [188]: Loading theme file: /tmp/.mount_curaw5CpbF/share/cura/resources/themes/cura-light/theme.json
2022-09-08 16:58:57,572 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [323]: Loaded theme /tmp/.mount_curaw5CpbF/share/cura/resources/themes/cura-light
2022-09-08 16:58:57,572 - INFO - [MainThread] UM.Qt.Bindings.Theme.load [324]: System's em size is 13px.
2022-09-08 16:58:57,573 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [323]: Loaded theme /tmp/.mount_curaw5CpbF/share/cura/resources/themes/cura-dark
2022-09-08 16:58:57,573 - INFO - [MainThread] UM.Qt.Bindings.Theme.load [324]: System's em size is 13px.
2022-09-08 16:58:57,846 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [344]: Creating post processing plugin view.
[... 271 lines cut ...]
2022-09-08 16:59:37,106 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for prusa_i3 took 0.06067204475402832 seconds.
2022-09-08 16:59:37,106 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2022-09-08 16:59:39,858 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [442]: file:///tmp/.mount_cura6nskEg/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-09-08 16:59:41,188 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2022-09-08 16:59:41,213 - INFO - [MainThread] cura.CuraApplication.closeApplication [623]: Close application
2022-09-08 16:59:41,213 - INFO - [MainThread] cura.CuraApplication._setLoadingHint [816]: Initializing Active Machine...
2022-09-08 16:59:41,224 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2022-09-08 16:59:41,225 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [96]: No active GlobalStack, set quality profile model as empty.
2022-09-08 16:59:41,225 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
[... cut ...]
2022-09-08 16:59:41,268 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [506]: Shutting down cura
2022-09-08 16:59:41,274 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [359]: Attempting to kill the engine process
2022-09-08 16:59:41,274 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [365]: Killing engine process
2022-09-08 16:59:41,275 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [368]: Engine process is killed. Received return code -15

Aaron7317, 2022-09-09:
One note: The file manager on the latest version 4 release works fine. This seems to be only an issue with version 5. Also I tried running version 5 natively and through Xwayland and both had the problem.

MariMakes, 2022-09-19:
Hey @Aaron7317,

Can you let us know if extracting the AppImage and running from the extracted folder is working for you?
If it doesn't work for you, we'll have someone from the team look at your issue. 

Thank you 😄

JudgeGregg, 2022-09-20:
Hello, it looks like a duplicate of #12266 to me ?
I had the very same problem on Arch with AppImage-5.1.1-linux (not modern), and installing `xdg-desktop-portal-gtk xdg-desktop-portal` fixed it.

Before doing that, I tried running the app from the extracted filesystem (./Ultimaker-Cura) and the file manager still did not open.
Thank you for your work on Cura :)

Aaron7317, 2022-09-20:
Thanks for the tips. Unfortunately, neither of them seemed to work. I already had xdg-desktop-portal-gtk and xdg-desktop-portal installed. When I extracted the AppImage and ran it from there, I had the same issue as well. Once again thanks for helping out :)

MariMakes, 2022-11-08:
Hey @Aaron7317,

Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12266
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12266

closed 2022-11-08 by MariMakes