Ultimaker/Cura #13250: NetworkX module no longer bundled with Cura 5.1, causing issues with Mesh Tools plugin

fieldOfView, 2022-09-09:
### Application Version

5.1.0

### Platform

any

### Printer

any

### Reproduction steps

Prerequisite: Cura 5.1 with MeshTools plugin installed from Marketplace.

1. load a model
2. Use Mesh Tools -> Fix simple holes...

### Actual results

Crash with the following traceback:
```
Traceback (most recent call last):
File "C:\Users\XXX\AppData\Roaming\cura\5.1\plugins\MeshTools\MeshTools\MeshTools.py", line 333, in fixSimpleHolesForMeshes
success = tri_node.fill_holes()
File "trimesh\base.py", line 1801, in fill_holes
return repair.fill_holes(self)
File "trimesh\repair.py", line 263, in fill_holes
g = nx.from_edgelist(
File "trimesh\exceptions.py", line 28, in __getattribute__
raise super(ExceptionModule, self).__getattribute__('exc')
File "trimesh\repair.py", line 19, in <module>
import networkx as nx
ModuleNotFoundError: No module named 'networkx' 

### Expected results

No crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

In Cura 5.1, networkx was removed as a bundled module. This module is a soft requirement for the trimesh module. Removing it from Cura installs has crippled the trimesh functionality.

I can start bundling networkx inside the Mesh Tools plugin, but I would argue that removing the networkx module from Cura without further notice is bound to introduce issues for other uses of trimesh, so putting it back in should be considered.

fieldOfView, 2022-09-14:
Can you shine some light on this @rburema? I believe the build-process is on your plate currently.

fieldOfView, 2022-09-14:
For details, see https://www.reddit.com/r/Cura/comments/x9gx0l/network_x_error_for_mesh_tools/

rburema, 2022-09-14:
Hi @fieldOfView 

Yeah, we saw the same error happening via Sentry, though it didn't specifically trace back to the plugin in that instance.

We've made PR's
https://github.com/Ultimaker/Uranium/pull/851
https://github.com/Ultimaker/Cura/pull/13288
to address this.

There should be enough time before 5.2 to merge that for that version at least.

I'm not sure why you think the build-process is on my plate specifically though. If anyone it's @jellespijker (but it really _should_ be all of us, until we can have a proper DevOps'er again).

fieldOfView, 2022-09-14:
Ha, I totally mixed up the two of you! (but it worked ;-) )

jellespijker, 2022-09-15:
> until we can have a proper DevOps'er again

:thumb_against_forehead_emoticon: not it!

MariMakes, 2022-09-15:
[image]

Thanks for the report @fieldOfView 
The issue seems to be resolved in the tickets that @rburema linked.
I'll be closing this issue.

closed 2022-09-15 by MariMakes