Ultimaker/Cura #13251: Error de corte.

Rubiodelta2, 2022-09-09:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Cuando le doy a segmentar a un archivo mi sala la advertencia de error de corte!!
[image]

### Actual results

[image]

### Expected results

Que debo hacer para solucionar el problema? Gracias.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-09-09:
Thanks for the report.
MS 3D Builder is bundled with Windows.  Open that file in 3D Builder and then on the toolbar click on "Insert".
The app has an Auto-Check function and you may be informed of errors in the model.  Click on the dialog to make the repairs and then save the file (STL, 3MF, etc) under another name.
Looking at your preview I can see red marks on the top edge of the model.  They indicate either Flipped Normals or gaps in the mesh.
There is a bug in Cura that can cause it to not slice a model with no errors.  A possible work-around is to move and/or rotate the model on the Cura build plate.
Let us know how you make out.  Your report might be a duplicate of others.

Rubiodelta2, 2022-09-09:
Fixed by moving the object around. Thanks.

GregValiant, 2022-09-09:
Alright.  I'll mark this as a duplicate.