Ultimaker/Cura #13252: error has occured while slicig

T0biasCZe, 2022-09-09:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1) opened a model
2) rotated the model so its easier to print
3) click slice
4) cura throws error after +- 10 seconds of slicing

(the slicing works when the model is not rotated)

### Actual results

cura doesnt slice the file and throws error

### Expected results

cura slices the file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9537687/cura.log)
[CE3_print fixedr.zip](https://github.com/Ultimaker/Cura/files/9537690/CE3_print.fixedr.zip)

GregValiant, 2022-09-09:
Thanks for the report.
You should have received the "Model is not watertight and may not print correctly" warning.  I know your file name says "fixed" but I get this in Cura...
[image]

The purple areas are errors.

This is after repair:
[image]

So I don't see a bug here.  The model has errors and though it might slice in one orientation, Cura hangs when the slicing is done from a different angle.

GregValiant, 2022-09-09:
I used MS 3D Builder.  It comes with Windows.

closed 2022-09-09 by T0biasCZe