Ultimaker/Cura #13253: Abort G-code

robertmartin, 2022-09-09:
### Is your feature request related to a problem?

I'm always frustrated when I click the Abort button and the bed and hot end do  not move where I would like them to.

### Describe the solution you'd like

In Machine Settings currently there is a Start G-code and End G-code box. There should be an Abort G-code box. When a print is aborted via the Abort button, the G-code in this box (if present) should override the default G-code and run for the user.

### Describe alternatives you've considered

The alternatives I have considered are, none really.

### Affected users and/or printers

It will effect all users and all printers.

GregValiant, 2022-09-10:
It doesn't work that way.  The "Abort" function is in the firmware and you can't do it in the gcode.  Gcode is a static file that simply gets read line-by-line from beginning to end by the printer/processor.  There is no way that the printer can be told "Wait...jump over to this section and run this code.".
You can send commands via the USB while a print is running off the SD card.  You would need to have a computer program to talk to the printer.  Something like Octoprint or Pronterface might be able to do it.  It won't be instantaneous though.  Your commands would start at the back of the buffer queue which is usually about 8 commands.

robertmartin, 2022-09-12:
I think you may have misunderstood. When the user clicks the abort button
in Cura, printer stops and the is queue empty, then the "on" (after) abort
custom G-code should run.

In my use case, I would like to move the print bed forward for removal and
cleaning. I also move the z-axis up for nozzle inspections. Currently I
have to do this manually every time.

GregValiant, 2022-09-12:
So you are talking about printing via the USB?  I did assume you meant the LCD button.
#13254 is also asking for changes to the USB interface.  Your request and that other one are things that would need to be handled by members of the community.  Ultimaker printers don't use USB printing and so Ultimaker will not be paying their people just so I can use it on my Ender.  If it works, it works and if it doesn't, it doesn't.  I think that if we want changes, then we need to talk a community member into it.  USB printing has been unmaintained for about 5 years and I don't see anybody jumping up to move it forward.