Ultimaker/Cura #13257: Slicing failed with unexpected error

RAEROVIA, 2022-09-10:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

Imported Object
Clicked Slice

### Actual results

Slicing Failed
Slicing failed with unexpected error

### Expected results

Slice Successful

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Bow 1.zip](https://github.com/Ultimaker/Cura/files/9539251/CE3_Bow.1.zip)

GregValiant, 2022-09-10:
Thanks for the report.
You can see in the image below that the model has a lot of errors as indicated by the red and purple areas.  It won't slice until it is repaired or redesigned.  Unless you shut it off - that warning dialog should have come up when you opened this model in Cura.
[image]

I'll remove the bug label as this doesn't appear to be a Cura issue.  I'll leave it open so you can respond.

Here you can see that it slices after being repaired.
[image]

MariMakes, 2022-11-10:
Hey @RAEROVIA,

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. 

Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2022-11-10 by MariMakes