Ultimaker/Cura #13258: Starting gcode does not save until another control has focus

EasyG0ing1, 2022-09-10:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Anycubic Mega X

### Reproduction steps

This only happens in the Windows version:

Go into Settings / Printer / Manage Printers
Select a printer then click on Machine Settings
Change the gcode, and without clicking anywhere else in the window, close the window using the X in the title bar

### Actual results

When you go back into Machine Settings you will notice that the new starting gcode was not committed.

### Expected results

Modified starting gcode should be committed once it is placed into the text area.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This feels like the text area that holds the gcode is set to possibly commit any changes when it loses focus, but it would seem that simply closing the window using the X button in the windows title bar doesn't trigger the controls lost focus event. If you first click into another control in the window, THEN close the window, the changes are committed.

GregValiant, 2022-09-10:
Thanks for the report.
This happened when the switch was made from Qt5 to Qt6 controls.  I don't know if it's being worked on or not.  Maybe @MariMakes has more information on where this is.
This behavior has been noted before so I'll mark this as a duplicate.  It is certainly annoying.

EasyG0ing1, 2022-09-10:
> This behavior has been noted before so I'll mark this as a duplicate #11941 and #12422. It is certainly annoying.

It does work just fine in MacOS ... and I assume that the same code is compiled for both operating systems? If that is the case, then perhaps it is a bug in QT6 that manifests from the Windows compiler ... but having no experience with qt, I'm just throwing darts in the dark. 
☺

GregValiant, 2022-09-10:
I'm a VB guy and it is an easy fix there ("on-change" instead of "lost focus").  Implementing the Qt6 controls from Python appears to be more of a challenge.
It looks like there have been a couple of other popular issues that have come up on the Windows side.  The "won't slice when centered" bug has come up a lot.

EasyG0ing1, 2022-09-10:
VB? .... as in ... Visual Basic? I think it's been maybe 20 years since Ive messed with visual basic. Im into Java these days. I looked at qt after reading your first response ... seems like it wants to be a whole lot of things. They sure do make it sound like the silver bullet of app development... Is this how people make GUIs with Python (a scripting language at its core)?

GregValiant, 2022-09-10:
I guess.  Cura gets ported to Windows, Mac, Linux and I think Chrome and a couple of others so a sort of universal set of controls sort of makes sense.
I hate "Universal" because you know that a cutting torch and a sledge hammer will be required and I hate "Ambidextrous" because it means it won't work either way.
There was a comment around here that Cura has outgrown Python.  I'm an amateur hack so It isn't something I can do anything about, but I've been thinking about that.

EasyG0ing1, 2022-09-10:
I've always seen Python as a scripting language, not a language for writing apps. Apparently, it has evolved over the decades, but it is still a single-threaded language and it doesn't even have true array support ... and pretty much everything it does has been the result of spit and duct tape patching a language together to get functionality from it. The odd thing about it is that it is the most used language in the world. Java is second followed by C/C++ ... and that always blows my mind ... a scripting language ... the most popular ... I can't fathom why that is the case.

nallath, 2022-09-15:
None of this had anything to do with python ;) 

You can see the fix I made here: https://github.com/Ultimaker/Cura/commit/8e41a03c58828dae799d32314a5900582c01c347 

It also happened on Linux and on Mac. The issue was basically that we only updated the setting when you lose focus. For some magical reason, it didn't consider closing a window as something that would make you lose focus.

EasyG0ing1, 2022-09-16:
> None of this had anything to do with python ;)
> 
> You can see the fix I made here: [8e41a03](https://github.com/Ultimaker/Cura/commit/8e41a03c58828dae799d32314a5900582c01c347)
> 
> It also happened on Linux and on Mac. The issue was basically that we only updated the setting when you lose focus. For some magical reason, it didn't consider closing a window as something that would make you lose focus.

I think that is exactly what I said in my opening statement in this issue ;-)