Ultimaker/Cura #13263: Crash when I press open file

svsdval, 2022-09-11:
### Application Version

5.1.0

### Platform

Debian GNU/Linux

### Printer

Custom

### Reproduction steps

I only press the Open File button

### Actual results

Cura crash with this output:

(Ultimaker-Cura:5652): Gtk-WARNING **: 16:05:53.603: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/gnome/16x16/status/image-missing.png: Нераспознанный формат файла изображения (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/gnome/16x16/status/image-missing.png: Нераспознанный формат файла изображения (gdk-pixbuf-error-quark, 3)
Aborted

### Expected results

Open any file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9542127/cura.log)

svsdval, 2022-09-11:
previous version (4.13) works fine, 5.0 also cant open file (open dialog file not opened) but doesnt crash

MariMakes, 2022-09-23:
Hey @svsdval,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm seeing some non-latin characters in your logs. 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/9644
We have a ticket on our backlog with the intent to improve this behavior.

A work around could be to change your path of the temp folder:

Search for environment variables in the windows search bar.
This will open system properties and there you can click on Environment Variables.
Edit the TEMP variable.
Change the Variable value to something like C:\Temp
Here you can read more about modify environment variables: https://docs.oracle.com/en/database/oracle/machine-learning/oml4r/1.5.1/oread/creating-and-modifying-environment-variables-on-windows.html#GUID-DD6F9982-60D5-48F6-8270-A27EC53807D0

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/9644

closed 2022-09-23 by MariMakes

svsdval, 2022-09-23:
This is on linux, and no non-latin characters in path's;

[cura.log](https://github.com/Ultimaker/Cura/files/9635755/cura.log)

[vars.txt](https://github.com/Ultimaker/Cura/files/9635733/vars.txt)

JohnCalin, 2022-09-27:
Here's a me too.

Please re-open this issue, it is not a duplicate of the issue mentioned.  It really has nothing in common with that issue at all, so whatever criteria was used to mark it as a duplicate, review your processes.

OP is on Debian linux, I am on Gentoo linux.
Not Windows, no unusual characters in the username, and the crash really has nothing to do with opening STL files.
I am able to open STL files without issue by using drag-and-drop, but as the OP described Cura bails out when it tries to display a file dialog.  I get the same behavior whether I am opening a new file or saving - anything that tries to show a file dialog.

JohnCalin, 2022-09-27:
For reference, here's my bail out message:

(Ultimaker-Cura:27538): Gtk-WARNING **: 13:38:20.028: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltimaDPZcha/AppRun: line 16: 27538 Aborted                 $scriptdir/Ultimaker-Cura "$@"

JohnCalin, 2022-10-11:
@svsdval I opened my own ticket for this issue, and it was found to be a dup of https://github.com/Ultimaker/Cura/issues/12266
There are some possible work-arounds listed there that may help you.

svsdval, 2022-10-11:
> @svsdval I opened my own ticket for this issue, and it was found to be a dup of #12266 There are some possible work-arounds listed there that may help you.

Thanks a lot ! ;)