Ultimaker/Cura #13265: Wall Line Count for top and bottom layers

Alkerion, 2022-09-11:
### Is your feature request related to a problem?

Actually it's not possible to set the wall line count for the top and bottom layers if we don't want to see them.

Such a feature could allow to have a better surface visual and sometimes perhaps strengthen the part too.

### Describe the solution you'd like

Been able to set the wall line counts for the top and bottom layers :

Bottom Wall Line Count : 0 to "Wall Line Count" value
Bottom Wall Line Layers : 0 to "Bottom Layers" value
Top Wall Line Count : 0 to "Wall Line Count" value
Top Wall Line Layers : 0 to "Top Layers" value

### Describe alternatives you've considered

Non

### Affected users and/or printers

All

Piezoid, 2022-09-11:
See #13080 for something similar. smartavionics' fork have it implemented.

IzzyMan12078, 2023-02-01:
This would be great