Ultimaker/Cura #13266: Even and Odd Wall Line Count for top and bottom layers

Alkerion, 2022-09-11:
### Is your feature request related to a problem?

Sometime the top/bottom layers to wall joint is't strong enough due to the fact that wall line count is the same from first to last layer.

I think that if the layers and the wall are more intricate it will reinforce the top/bottom structure.

Been able to intricate the layers and walls could probably helps to strengthen the structure of some (if not all) objects.

### Describe the solution you'd like

To have a simple option to automatically alternate the wall line count to intricate them with layers.

Example : 

Wall Line Count : 4 walls
Bottom Layers : 4

New option :
Minimum Wall Line Count : 2

Result to :
1st layer : 2 walls
2nd : 3 walls
3rd : 2 walls
4th : 3 walls
5th to ... : 4 walls

### Describe alternatives you've considered

Been able to set Even and Odd Wall Line Count for top and bottom layers

Even Wall Line Count
Odd Wall Line Count

### Affected users and/or printers

All

### Additional information & file uploads

[image]

Piezoid, 2022-09-11:
"Alternate Extra Wall" is already doing something very similar, albeit with a variation over a single wall width.

Alkerion, 2022-09-11:
Thanks, I wasn't aware of this setting.

Sadly it's applied on the whole part, not only top/bottom.

I'll perform a few tests.